checkra1n/PongoOS #6: Getting [Errno 32] Pipe error on iPhone 6+

ForceBru, 2020-03-06:
So I've built the `example` module and now trying to load it somehow.

First, is there _any_ documentation on how to use this thing?

Since Checkra1n operates on devices in DFU mode, I put my iPhone into DFU mode as well. Then executed lines from `scripts/module_load_boot.py` line-by-line (BTW, my `idProduct != 0x4141`, so maybe no need to hard-code it), but it errors out here:

https://github.com/checkra1n/pongoOS/blob/1dc577ff3e139755545b94123b9a83e3e8c93bb2/scripts/module_load_boot.py#L31

Traceback:

```
>>> dev.ctrl_transfer(0x21, 2, 0, 0, 0)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/usb/core.py", line 1043, in ctrl_transfer
    self.__get_timeout(timeout))
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/usb/backend/libusb1.py", line 883, in ctrl_transfer
    timeout))
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/usb/backend/libusb1.py", line 595, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBError: [Errno 32] Pipe error
```

So, how can I solve this? Checkra1n itself runs fine, `pyusb` detects the device, iTunes detects it too, so there are no issues with the USB connection.

woachk, 2020-03-06:
Hello,

Please test USB for Apple A9 or later only or now... it's broken on A7 and A8 for an oddball cache coherency issue that isn't quite identified yet.

ForceBru, 2020-03-06:
@woachk, so is there no way to upload pongoOS and/or my module onto my iPhone yet? If so, how come Checkra1n is able to do this?

redoxinc, 2020-03-07:
checkra1n is not working on my A7 device so the warning seems to be true

ForceBru, 2020-03-08:
Okay, so apparently that's just me being stupid and not actually running pongoOS, but the `0x21` control [is pongoOS-specific](https://github.com/checkra1n/BugTracker/issues/1009#issuecomment-596125153), so it won't work if pongoOS isn't running.

However, it looks like it doesn't work on A8 yet anyway, so waiting for a fix

kpwn, 2020-03-10:
pongoOS is to be booted with checkra1n

Genanik, 2020-03-11:
I also faced this problem on my iPhone 6. If I execute the command twice, the pongoOS will panic, and the error code is 0x73656e64. I hope this can help.

mjd0, 2020-04-07:
LICENSE.md

mjd0, 2020-04-07:
module_load_boot.py

mjd0, 2020-04-07:
module_load_boot.py

justtryingthingsout, 2020-09-25:
Additional information: 

1. when running upload_data.py, it seems to actually be uploading something (loading circle in terminal), then suddenly it disappears, and when you terminate it, the `pongoOS USB Device` still shows up in System Information and python, but when you try and run the command again, it says `usb.core.USBError: [Errno 19] No such device (it may have been disconnected)`.

2. most of the time `pongoOS USB Device` doesn't actually show up, it just shows as `USB Device`, with only a Location ID, and it sometimes disconnects and reconnects to the mac, sometimes disconnecting entirely, only being access the shell with serial (serial works even if System Information shows the device not connected at all)

3. other times, you can be extremely lucky and successfully get the iManufacturer, iProduct, iSerialNumber, etc from pyusb (normally you get `Error Accessing String` for all of those). This still however, runs into the same issues of the scripts breaking.

4. issue_cmd.py seems to be the only script I found so far that causes the `panic: USB BUG: 0x73656e642034` bug, others either work fine (fetch_stdout.py works fine as long as you input something first on serial), or freeze up like upload_data.py.

woachk, 2020-10-31:
Fixed in the latest release. Please reopen the issue if bugs are present.

closed 2020-10-31 by woachk