checkra1n/PongoOS #8: Latest version always boots into Pongo shell

ForceBru, 2020-03-23:
PongoOS from commit 4ebe9c5 always boots into Pongo shell, no matter the arguments to `checkra1n`. I'm running it as follows:

    /Users/forcebru/Desktop/Jailbreak/checkra1n/checkra1n.app/Contents/MacOS/checkra1n -k /Users/forcebru/Desktop/Jailbreak/checkra1n/pongoOS/build/Pongo.bin

Setting `CHECKRA1N_PONGO=0` before running the command doesn't help.

Setup:

* iPhone 6+
* iOS 12.4
* Checkra1n beta 0.9.8.2
* pongoOS 4ebe9c5

This doesn't happen with pongoOS shipped with Checkra1n (1.0-641...)

woachk, 2020-03-23:
Hello, That's not an issue but by design. It's because the kernel patchfinder is _not_ included on the public pongoOS source tree.

You can bundle an image that supports automatic booting with:
`	echo 6175746F626F6F740000200000000000 | xxd -ps -r | cat $(BUILD)/Pongo.bin /dev/stdin $(BUILD)/checkra1n-kpf-pongo > $@
`

Extracting the pongoOS KPF is left as an exercise for the reader.

closed 2020-03-23 by woachk

ForceBru, 2020-03-23:
> Extracting the pongoOS KPF is left as an exercise for the reader.

So this repo is completely useless to a regular user who wants to try out the new pongoOS, right? I have no idea what a "kernel patchfinder" is, let alone how to use it, what it does or where to get it, so am I immediately out of play here?

All I wanted was to write my own simple module for pongoOS, just for fun, but then it turned out that [USB doesn't work](https://github.com/checkra1n/pongoOS/issues/6) (it _still_ doesn't), and now, even if USB did work, I wouldn't've been able to `autoboot` anyway?!

woachk, 2020-03-23:
@ForceBru USB works on Apple A9 devices and later, if you're on A8 and earlier you'll have to wait a bit more.

The kernel patchfinder is to patch out signature checks in the kernel among other things. And yeah, this repository isn't really geared towards regular users.

ForceBru, 2020-03-23:
@woachk right, I'd better wait for proper USB support then. I wish I knew USB to help out with it here...
Thanks for your replies!