checkra1n/PongoOS #9: PongoOS Flash-NOR / SPI driver

rickmark, 2020-03-26:
When booted to pongoOS over iBoot we lose read/write access to persistent storage.

Pongo ought to provide access to these methods, as they are key to the setup of the next phase boot loader

mjd0, 2020-04-07:
> When booted to pongoOS over iBoot we lose read/write access to persistent storage.
> 
> 
> 
> Pongo ought to provide access to these methods, as they are key to the setup of the next phase boot loader

rickmark, 2020-04-08:
> When booted to pongoOS over iBoot we lose read/write access to persistent storage.

> Pongo ought to provide access to these methods, as they are key to the setup of the next phase boot loader

There has been some discussion lately of doing exactly this, by jumping back to the SecureROM version of iBoot to call the functions bdev_read / bdev_write / bdev_erase

tjkr0wn, 2020-10-30:
Sounds like an interesting idea. Off the bat, I can think of two ways of implementing this. Either passing an iBoot and extracting the fragments of the driver and reconstructing and sending it to pongoOS, or having to rewrite the driver completely similar to what Brandon Azad did for KTRW (now reimplemented in pongoOS I believe.) What do you think?

rickmark, 2020-10-31:
Best option is a net new implantation as who knows what the side effects of iBoot code can be.  If I had time I’d be starting with the SPI first, and later moving to the NAND block layer... I think it works by making requests to ANS2 to drop a block into DRAM via DART

tjkr0wn, 2020-10-31:
> Best option is a net new implantation as who knows what the side effects of iBoot code can be.  If I had time I’d be starting with the SPI first, and later moving to the NAND block layer... I think it works by making requests to ANS2 to drop a block into DRAM via DART
> 
> Get Outlook for iOS<https://aka.ms/o0ukef>
> From: Tarek Joumaa <notifications@github.com>
> Sent: Friday, October 30, 2020 11:58:31 AM
> To: checkra1n/pongoOS <pongoOS@noreply.github.com>
> Cc: Rick Mark <rickmark@outlook.com>; Author <author@noreply.github.com>
> 
> 
> Sounds like an interesting idea. Off the bat, I can think of two ways of implementing this. Either passing an iBoot and extracting the fragments of the driver and reconstructing and sending it to pongoOS, or having to rewrite the driver completely similar to what Brandon Azad did for KTRW (now reimplemented in pongoOS I believe.) What do you think?

Sounds like a worthwhile invest in time as the idea of pongoOS is to overtime turn into an iOS version of Clover afaik.

rickmark, 2020-11-05:
For reasons of forensics / investigation of my devices that seem to have some form of malware persistence, I've started to undertake a read-only version of this work.  (I have iDevices that are hiding data in alternate NVMe namespaces).

Feel free to follow along to this: https://github.com/t8012/pongo-flash

tjkr0wn, 2020-11-05:
That’s a bit unfortunate. I currently only have a t8015 for testing purposes. Is reversing there a good idea or should I try to focus on the t8012 chip even though I don’t have it.

tjkr0wn, 2020-11-06:
@rickmark I have forked the repo and I'm gonna try to add some reversing/research notes related to the development of the driver there. It's under the same name.

rickmark, 2020-11-06:
The ipwndfu for t8012 probably works on other chips (for booting without the NAND).  @h0m3us3r, @aunali1 and @mrmacarm?

As for storage - since there is no device or driver concept I had to start the work on the internal version of Pongo, and it will probably need some API / design revisions by the team before we want to expose it for dependencies.  Rev0 will probably just be SPI/NOR (so just SCfg, effaceable, iBoot and the like)

tjkr0wn, 2020-11-06:
@rickmark I'd be happy to help out with the driver in any way I can! My current objective is to get a working PoC functional enough to at least read bytes from NOR