checkra1n/PongoOS #10: (question) how to modify or erase file

filipef101, 2020-04-06:
Trying to see if somehow can jailbreak on 12.1.
Want to start modifying something in fs.
any basic example doing that?

rickmark, 2020-04-08:
In order to use the filesystem, you'll need something far higher up the boot chain than pongoOS.  Pongo is a shim for the next phase and doesn't have the required Flash and filesystem drivers.  If the device is an X or prior try checkra1n which does enable this by booting a jailbroken XNU kernel.

filipef101, 2020-04-14:
@rickmark Is that the "terminal" that boots up?

rickmark, 2020-04-14:
Yes it is.  PongoOS is this repo.  It has a module system so hopefully people write some block device/filesystem modules to provide what you need.

Siguza, 2020-04-14:
Probably more reasonable to just boot a Linux ramdisk and use existing drivers tbh...

closed 2020-06-02 by filipef101

Clindbergh, 2020-10-17:
> 
> 
> Probably more reasonable to just boot a Linux ramdisk and use existing drivers tbh...

@Siguza What do you think is the easiest way to get file system access with a minimal image? I have a ios 13.5 iphone8,1 that doesn't boot and with an ios 14 update already initiated but booting into recovery mode. Possibly sep-firmware.img4 is already gone, hopefully not. Filesystem access to backup files before updating is what I'm going for here.

rickmark, 2020-10-17:
So on "minimal images" I'd say two roads:

1) Use `sandcastle` since it's already a linux image
2) Boot the Apple restore/update ramdisk because this is purpose built for this (you'd have to replace `asr` with something useful)

danieltroger, 2020-10-18:
@rickmark Re: the second option

He already has a customized ramdisk with telnet and bunch of utilities on it. The question is how he'd load devicetree, kernelcache and trustcache with pongoOS. I know there's the ramdisk command, but just sending a ramdisk and the ramdisk command isn't enough to boot it iirc.