checkra1n/PongoOS #17: Unable to interact with pongoOS on Apple TV4

NewDwarf, 2020-05-01:
On attempt to communicate with pongoOS on Apple TV4 I get a timeout.

```
bash-3.2$ python issue_cmd.py ?
Traceback (most recent call last):
  File "issue_cmd.py", line 31, in <module>
    dev.ctrl_transfer(0x21, 3, 0, 0, sys.argv[1] + "\n")
  File "/usr/local/lib/python2.7/site-packages/usb/core.py", line 1043, in ctrl_transfer
    self.__get_timeout(timeout))
  File "/usr/local/lib/python2.7/site-packages/usb/backend/libusb1.py", line 883, in ctrl_transfer
    timeout))
  File "/usr/local/lib/python2.7/site-packages/usb/backend/libusb1.py", line 595, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBError: [Errno 60] Operation timed out
```
The last messages on the TV screen.
...
Pongo shell requested, stopping here!
Enabling USB
Done!
pongoOS> set xnu boot arg cmdline to: [rootdev=md0]
pongoOS>

At the same time I am able to interact with pongoOS on iPhone.

Version: # Checkra1n beta 0.10.1

Siguza, 2020-05-03:
The AppleTV 4 uses an A8 chip, which makes this more than like a duplicate of #6.  
In summary, A7 and A8 have weird USB controllers that we don't fully support yet.

closed 2020-05-03 by Siguza

tmm1, 2023-04-29:
@woachk closed #6 as fixed, but that wasn't enough to get Apple TV support working. can this issue be re-opened?