checkra1n/PongoOS #19: Controlling of the kernel patch finder

NewDwarf, 2020-05-02:
Is it possible to control kernel patch finder by the 'kpf_flags' command so that to enable/disable sandbox patch/es?
It is interesting to play with the Siguza's sandbox escape bug in the sandboxed environment.

Siguza, 2020-05-03:
It's currently "all or nothing", but I suppose this makes sense as a feature request.

vip1t1413, 2020-06-11:
> Is it possible to control kernel patch finder by the 'kpf_flags' command so that to enable/disable sandbox patch/es?
> 
> It is interesting to play with the Siguza's sandbox escape bug in the sandboxed environment.

RPwnage, 2021-03-18:
I’ll try to work this out and create a pull request

rllbe, 2021-12-05:
I second this, as for now some apps are clever enough to detect jailbreak via sandbox behavior differences(for example SYS_open to "/var/mobile" would success, and SYS_csops would always have CS_PLATFORM_BINARY flag set when booted by checkra1n, and more). Pairing with SYS_access detection to jailbreak bootstrap files makes those apps invincible to any jailbreak detection bypass tweaks. Something made to activate these changes for certain processes on the go instead of just hooking them all would be nice IMO. 

By the way are there any ways to load a pongo module without breaking sandbox or loading kpf at all? I tried a couple of ways to disable kpf including disabling kpf at all(boot via Pongo.bin) or commenting out flag changes in amfi_execve_hook but either the module wouldn't load or CS_PLATFORM_BINARY still got set.

JQ555888, 2022-04-21:
Following this issue as well. 

I was rather surprised to find out checkra1n's sandbox behaviour allows read access to not only /var but also the entirety of /var/mobile. Granted I'm not the most knowledgeable on this and the team probably has a good reason to do this other loading the bootstrap onto the device, but from a security and usability standpoint I'm not sure this is the most sensible kinda implementation especially when you compare it to the sandbox behaviour on other app based jailbreaks, let alone there's detection implications as well. I wasn't paying a lot of attention to this issue until whilst I was using a-Shell/LibTerm oe and noticed that App Store apps had no problem reading my private keys under ~/.ssh and ~/.local which turned out quite concerning for an app directly downloaded from an official source. Being able to adjust the sandbox patches/an option to make it a bit more strict at boot time would be a nice addition.