checkra1n/PongoOS #22: pongoOS sometimes hangs at "Booting" after module load and bootx

jsherman212, 2020-08-11:
While writing a pongoOS module, I noticed the device would sometimes hang at "Booting" after `bootx` was issued after a module was loaded:
[image]

At first I thought it was something my module was doing, but this issue is present on a module that just registers its own preboot hook that invokes checkra1n's preboot hook to run its KPF.

This happens around 10% of the time on an iPhone X running iOS 13.3.1, an iPhone 8 running iOS 13.6, and an iPhone SE (original, 2016) running iOS 13.6. The only device I could not get this issue to happen on is an iPhone 7 running iOS 13.1.2. 

I attached the barebones module which causes the problem. I use libusb 1.0.23 to communicate with the device, and I compile this on macOS 10.15.4. To compile, unzip to some directory, `cd` to that directory, and `make`. Then: `loader/loader module/module` and boot the device to a pongo shell. checkra1n version is beta 0.10.2.

[pongo_hanger.zip](https://github.com/checkra1n/pongoOS/files/5058364/pongo_hanger.zip)

woachk, 2020-08-11:
Hello,

could you add -v to the boot arguments and take a picture please to see where it hangs in XNU?

Thank you,

jsherman212, 2020-08-11:
XNU never actually boots, it just hangs at "Booting", so even with -v nothing is shown

funcoleto, 2021-08-08:
> XNU never actually boots, it just hangs at "Booting", so even with -v nothing is shown

please, could you send me the pongo.bin version 1.2.1?

Equalis-programming, 2024-02-19:
Got the same issue with Iphone 8 and iOS 16.7.4.
I first launched palera1n from an intel mac with the cf flags then only the f flag before my phone entered pongoOS
I couldn't get out of that state, wich seems like an infinite loop for me before my phone ran out of battery.
I retried but only got the same issue
I think it's a problem with Palerain's cf flags