checkra1n/PongoOS #24: Can’t load raw ibss

ak99004242, 2020-08-19:
I m trying to load raw ibss from pongo shell. When I send raw ibss and bootr then it just hangs on a white screen doesn’t allow to send ibec. But if i send raw ibec From pongo shell it boots fine, but the problem is I want to load ibss first and then ibec as I can’t use go command properly with out. Please help. Thanks

Siguza, 2020-08-20:
I'm honestly amazed that iBEC even works, but I guess we do just barely enough patches for that.  
That said, two points:

- iBSS expects the device to be in the state the ROM put it in, which includes DRAM being not up yet. This is not feasible for us to support.
- Recovery mode starts the chain with an iBEC, so I really don't understand the need to load an iBSS.

ak99004242, 2020-08-20:
I have replicated the whole process on A9 device with eclipsa and it works fine since I can send ibss before ibec. If I use the same ibec to send commands by pongo I can’t use go command. It looks like there is something missing. My main aim was to boot ibec and send go command using A9x as eclipsa doesn’t support A9x and there is no ipwndfu for it. Is there anything that can be done please.

lightcode4, 2020-08-28:
[image]

Anyone knows what the issue might be ?? 
iPhone X 
Os ; 13.0
Checkra1n: all version didn't work

Liminxin123, 2020-10-24:
This is your apple cell phone X lead to confusion in the system files, many failed to jailbreak, only to replace the motherboard memory, such as a series of apple original accessories.

Liminxin123, 2020-10-24:
And if.Checkra1n jailbreak failed again, then you use Odyssey jailbreak your Apple X retest.

Liminxin123, 2020-10-24:
If you continue to use Checkra1n jailbreak, your iPhone will be directly black screen directly disabled, which is equivalent to a permanent sleep of the phone.

Liminxin123, 2020-10-24:
You belong to Checkra1n.FAUCheckra1nThe jailbreak file is missing or incomplete during the jailbreak process! ！ ！ ！

Mongs29, 2020-11-16:
> I m trying to load raw ibss from pongo shell. When I send raw ibss and bootr then it just hangs on a white screen doesn’t allow to send ibec. But if i send raw ibec From pongo shell it boots fine, [...]

did you success booting diags? this will be usefull for devices dont have pwndfu ..hope you respond 🙏