checkra1n/PongoOS #25: Can't boot pongoOS

Clindbergh, 2020-10-18:
I have an iphone 6s (N71mAP) with iOS 13.X that only booted into recovery mode after low storage. I tried to update with iTunes to iOS 14, but the update failed after about 1/3 (of progressbar during the update). I didn't try to restore from iTunes since I wanted to save data.

With [PyBoot](https://github.com/MatthewPierson/PyBoot)/[eclipsa](https://github.com/0x7ff/eclipsa) (as part of @danieltroger nice [telnet ramdisk](https://github.com/danieltroger/telnetd_ramdisk), thx!) I wasn't able to get into pwndfu mode. With [king](https://github.com/pgarba/King) I was more successful and `irecovery -s` confirmed the checkm8 pwned dfu mode.

I tried to boot pongo `/Applications/checkra1n.app/Contents/MacOS/checkra1n -p`, but the iPhone boots straight into recovery mode. I tried multiple times with two different Macs.

IIWare, 2020-10-25:
try `/Applications/checkra1n.app/Contents/MacOS/checkra1n -cp` the `c` is force cli mode and the `p` boots to pongoOS

Siguza, 2020-10-25:
@IIWare Pongo only boots if iBoot exits though.

@Clindbergh I guess you could try uploading a signed iBEC to recovery...

Clindbergh, 2020-10-28:
> @Clindbergh I guess you could try uploading a signed iBEC to recovery...

@Siguza: Could you elaborate how that would work? I think I tried it as [part](https://github.com/danieltroger/telnetd_ramdisk/blob/master/make_telnet_rd.php) of the telnet ramdisk.

How likely do you think is a Hardware (USB) defect? I am thinking about replacing the lightning board.

XP1, 2021-04-13:
I am in the same exact situation with my iPhone 6s (N71mAP) that failed to update because of error 14 (out of space). Running `checkra1n -cp` restarts the device from DFU into recovery.

See this thread on booting the images using irecovery by entering the commands manually. @ajpayne1 was able to create an SSH ramdisk. I'm still trying to figure that out...

iOS 13.5 / iPhone10,1 / Unable to load ramdisk #9:
https://github.com/danieltroger/telnetd_ramdisk/issues/9

Siguza, 2021-04-13:
So, we patch iBoot in a way that when it hands off execution to the next stage (usually the kernel), it will jump to Pongo instead. The problem that you're running into is that your file system is broken and iBoot cannot load the kernel, and it will thus never jump to Pongo.

But once it drops to recovery, you should be able to take the iBEC from the latest IPSW, fetch a valid APTicket for it and upload it to recovery. This iBEC will never run, it's just needed to satisfy iBoot. You would then use the Pongo USB shell to replace the loaded iBEC with an extracted kernel, and attach a custom ramdisk.

amats211, 2023-03-09:
> So, we patch iBoot in a way that when it hands off execution to the next stage (usually the kernel), it will jump to Pongo instead. The problem that you're running into is that your file system is [...]

Hi sir, sorry for bumping an old issue. But  can you recommend a tutorial or write up on how to patch iBoot(ibss,ibec) maybe using IDA, to boot to the next stage? Thank you very much