checkra1n/PongoOS #26: how can I type in the pongoOS shell?

IIWare, 2020-10-28:
Hi, I would like to type commands in the pongoOS shell, can you please tell me any method at all to type in the pongoOS shell? thanks.

Siguza, 2020-10-28:
If you're on macOS:

```
cd scripts
make
./pongoterm
```

Otherwise there's `issue_cmd.py` and `fetch_stdout.py`.

IIWare, 2020-10-28:
so is it like this: `./pongoterm [command]` ?

Siguza, 2020-10-28:
No, you run `./pongoterm` and you get an interactive shell.  
Alternatively you can do `./pongoterm <<<'command'` and it will run that once it finds a device, then exit.

IIWare, 2020-10-31:
@Siguza thanks for helping me, but there is a problem, in mac os when I run ./pomgoterm it says this:
`USBControlTransfer: (iokit/usb) pipe has stalled, error needs to be cleared`
Do you have any solutions for me to fix the problem? thanks, and sorry for many questions.

woachk, 2020-10-31:
Hello,

Which SoC does your phone use? Did you also try the newest version as of today?

IIWare, 2020-10-31:
@woachk My phone has A10, yes I tried the newest version as of 31/10/2020.

DanTheMann15, 2020-11-11:
on Ubuntu, i can't seem to get any access to the pongo shell.

pongoterm is hopeless on linux, and attempting to use the script `issue_cmd.py` doesn't work,
it vomits on me with this when i attempt to run it:

```
daniel@DansPPC2011:~/Desktop/pongoOS/scripts$ ./issue_cmd.py
./issue_cmd.py: line 23: import: command not found
./issue_cmd.py: line 24: import: command not found
./issue_cmd.py: line 25: syntax error near unexpected token `('
./issue_cmd.py: line 25: `dev = usb.core.find(idVendor=0x05ac, idProduct=0x4141)'
```

i want to get SEP keys for the wiki, but until i figure this out i can't do it.

i have access to s8000, s8003, and t8010

matteyeux, 2020-11-11:
@DanTheMann15 it's a python script with no shebang, so you have to run with python like this : ` python3 ./issue_cmd.py`

DanTheMann15, 2020-11-11:
> @DanTheMann15 it's a python script with no shebang, so you have to run with python like this : ` python3 ./issue_cmd.py`

thanks for the response, i added the python3 before the command and also needed to get pyusb's usb module as well and it worked.

but i can't yet pwn sep on s8000/s8003, but at least i can do t8010.

keys coming soon once i get it working right, thanks!

cw2k, 2021-03-18:
I'm tried `fetch_stdout.py` on some arch linux.
Well you may run:
`sudo pip install  pyusb libusb`
to get the necessary dependencies installed.

However even then I encountered that nasty 
`usb.core.USBError: [Errno 5] Input/Output Error`
what is in real some permissions problem.
Here is how to solve it: 
https://stackoverflow.com/a/66694064/3135511

closed 2024-07-01 by cuj