checkra1n/PongoOS #27: USBControlTransfer error

Athja, 2020-10-28:
everytime i do "./pongoterm" it says "connected" then "USBControlTransfer: (iokit/usb) pipe has stalled, error needs to be cleared" and doesnt give a shell. Why is this happening?

Athja, 2020-10-28:
also after like a minute it randomly boots into ios

woachk, 2020-10-28:
Hello,

do you use a new pongoOS too?

Athja, 2020-10-28:
yes

IIWare, 2020-10-31:
I have the same issue too.

YCarlB, 2020-11-04:
I have the same issue too.

Siguza, 2020-11-04:
What version of PongoOS are you using? (Should be printed at the top of the screen on your device.)

IIWare, 2020-11-04:
@Siguza I will find out because I have the same issue.

IIWare, 2020-11-05:
@Siguza my version:
`pongoOS 1.3.0-2e7eb6bc (EL1)`
and I still have this issue: USBControlTransfer: (iokit/usb) pipe has stalled, error needs to be cleared

woachk, 2020-11-05:
@IIWare That's... way too old for this feature.

You can use ./checkra1n -p -k Pongo.bin with building a new pongoOS from this repo.

IIWare, 2020-11-05:
@woachk so I do /Applications/checkra1n.app/contents/MacOS/checkra1n -p -k <insert pongo.bin filepath here> ? or the Pongo.bin will just export?

Siguza, 2020-11-05:
```
git clone https://github.com/checkra1n/pongoOS.git
cd pongoOS
make
path/to/checkra1n -k build/Pongo.bin
```

IIWare, 2020-11-06:
@Siguza finally! this helped!

Athja, 2020-11-06:
yes this worked. thank you

closed 2020-11-06 by Athja