checkra1n/PongoOS #31: SEP is not pwned!

joshblah555, 2020-11-10:
Would it be possible to pwn sep on Apple TV 4/4K?

lilstevie, 2020-11-11:
I just checked on my Apple TV with UART to grab the pongo log and can confirm it pwns sep 

`#==================
#
# pongoOS 2.4.1-87fe5ec4 (EL1)
#
# https://checkra.in
#
#==================
Booted by: iBoot-6723.0.43
Built with: Clang 8.0.1 (Red Hat 8.0.1-1.module_el8.1.0+215+a01033fb)
Running on: t8011
[modload_macho:i] Attempting to load a module
[modload_macho:+] Loaded module checkra1n-kpf2-12.0,14.0

#==================
#
# checkra1n kpf 0.12.0
#
# Proudly written in nano
# (c) 2019-2020 Kim Jong Cracks
#
# This software is not for sale
# If you purchased this, please
# report the seller.
#
# Get it for free at https://checkra.in
#
#====  Made by  ===
# argp, axi0mx, danyl931, jaywalker, kirb, littlelailo, nitoTV
# never_released, nullpixel, pimskeks, qwertyoruiop, sbingner, siguza
#==== Thanks to ===
# haifisch, jndok, jonseals, xerub, lilstevie, psychotea, sferrini
# Cellebrite (ih8sn0w, cjori, ronyrus et al.)
#==================
Found old-style rdsk!
set xnu boot arg cmdline to: [rootdev=md0 serial=3]
successfully obtained SEPROM code execution
SEP payload ready to boot
`

joshblah555, 2020-11-11:
Interesting. This is what I got: 

`JTV@macOS:~$ pongoterm
[Connected]

#==================
#
# pongoOS 2.4.1-87fe5ec4 (EL1)
#
# https://checkra.in
#
#==================
Booted by: iBoot-6723.43.1
Built with: Clang 8.0.1 (Red Hat 8.0.1-1.module_el8.1.0+215+a01033fb)
Running on: t7000
[modload_macho:i] Attempting to load a module
[modload_macho:+] Loaded module checkra1n-kpf2-12.0,14.0

#==================
#
# checkra1n kpf 0.12.0
#
# Proudly written in nano
# (c) 2019-2020 Kim Jong Cracks
#
# This software is not for sale
# If you purchased this, please
# report the seller.
#
# Get it for free at https://checkra.in
#
#====  Made by  ===
# argp, axi0mx, danyl931, jaywalker, kirb, littlelailo, nitoTV
# never_released, nullpixel, pimskeks, qwertyoruiop, sbingner, siguza
#==== Thanks to ===
# haifisch, jndok, jonseals, xerub, lilstevie, psychotea, sferrini
# Cellebrite (ih8sn0w, cjori, ronyrus et al.)
#==================
Found old-style rdsk!
Pongo shell requested, stopping here!
pongoOS> set xnu boot arg cmdline to: [rootdev=md0]
pongoOS> sep auto
pongoOS> sep decrypt 5986741eddee9e141bb06313fd429647a829106148798d6208ed3a06a3c98ece31a7e9a8623b065a6b50490d0b997791                            
sep is not pwned!
pongoOS>

`

Unless I'm doing it wrong...

Siguza, 2020-11-11:
You're on A8.  
Only A10, A10X and T2 are supported at the moment. Support for A8(X) and A9(X) is planned for the future.

joshblah555, 2020-11-11:
Ah, thanks for clarifying!

DanTheMann15, 2020-11-11:
ah that i was about to post about pwning sep on A9 (s8003), but this answers my question, i hope A9 support gets added soon, thanks.

frankpanduh, 2024-03-29:
> You're on A8. Only A10, A10X and T2 are supported at the moment. Support for A8(X) and A9(X) is planned for the future.

Did A9(x) ever get added?