checkra1n/PongoOS #33: fdt command removed in commit 696c1ee5d11cf46ab60460d35176bea48fa69cfe

squishykid, 2020-11-22:
I am trying to run project sandcastle using version 0.12.1 of checkrain, which includes a recent version of pongoOS. The sandcastle start script fails due to "Unknown command: fdt". Looking at the git history, is seems the command was removed in commit 696c1ee5d11cf46ab60460d35176bea48fa69cfe. Could you please tell me a little about why this command was removed. Is there an alternative option?

squishykid, 2020-11-22:
Checkra1n beta 0.10.2 seems to include a pre-2.0 version of checkra1n, which I was able to use to get project sandcastle running.

squishykid, 2020-11-22:
It is also possible to use the checkra1n command line option `--override-pongo` to supply a pogo.bin file from e111affda94a0562843aba82ca40f0768631f116.

woachk, 2020-11-22:
Hello,

That code was mostly moved to https://github.com/checkra1n/pongoOS/tree/master/src/modules/linux, we'll have to include it again in the build system. It'll be done in the next release.

bnaambo, 2020-12-21:
short question. here on GitHub it says the lastest release is 2.4.1 however on my iPhone its 2.4.3 but still there is no fdt

woachk, 2020-12-31:
Should be fixed now.

closed 2020-12-31 by woachk

j4ix, 2021-01-07:
the latest version of checkra1n uses pongo 2.4.3 and fdt is not there, says it can't copy  image to device and phone says bad command, is there a way to manually build the update and instantiate it?

Siguza, 2021-01-07:
@j4ix If you're on macOS, a simple `make` is enough. On Linux you'll need at least Apple's ld64.  
But you can also just download the CI build artifacts... https://github.com/checkra1n/pongoOS/actions?query=workflow%3ACI

Just feed PongoConsolidated.bin to checkra1n with `-k`.

Tsumiicode, 2021-04-09:
Hi in Pongo os 2.4.4-f22064f9 fdt is gone again? because project sand castle isnt working

ashcrel, 2021-04-30:
@Tsumiicode unfortunately I don't think the fdt command has been readded ever since. I'm looking through the commits and if I'm not mistaken the command simply doesn't exist in the current code.

Siguza, 2021-04-30:
It's right here: https://github.com/checkra1n/pongoOS/blob/09d82c8efcc28fc09e3391f48f41cb7b79a00808/src/shell/linux.c#L41-L52  
Registered here: https://github.com/checkra1n/pongoOS/blob/09d82c8efcc28fc09e3391f48f41cb7b79a00808/src/shell/linux.c#L85-L89  
Called from here: https://github.com/checkra1n/pongoOS/blob/09d82c8efcc28fc09e3391f48f41cb7b79a00808/src/shell/main.c#L267-L268

You can also just run `help` in Pongo shell and see if `fdt` is listed or not.

ashcrel, 2021-04-30:
ahhh I'm new to this stuff, this is really helpful. thanks! I'll edit my other reply.