checkra1n/PongoOS #35: `issue_cmd.py` does not work with python 3.9

CRKatri, 2021-01-03:
When running `sudo python3 ./issue_cmd.py help` I get the output
```
❯ sudo python3 ./issue_cmd.py help              
Traceback (most recent call last):
  File "/usr/lib/python3.9/site-packages/usb/core.py", line 1055, in ctrl_transfer
    buff = util.create_buffer(data_or_wLength)
  File "/usr/lib/python3.9/site-packages/usb/util.py", line 164, in create_buffer
    return array.array('B', _dummy_s * length)
TypeError: can't multiply sequence by non-int of type 'str'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3.9/site-packages/usb/_interop.py", line 94, in as_array
    return array.array('B', data)
TypeError: cannot use a str to initialize an array with typecode 'B'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/cameron/Documents/pongoOS/scripts/./issue_cmd.py", line 31, in <module>
    dev.ctrl_transfer(0x21, 3, 0, 0, sys.argv[1] + "\n")
  File "/usr/lib/python3.9/site-packages/usb/core.py", line 1057, in ctrl_transfer
    buff = _interop.as_array(data_or_wLength)
  File "/usr/lib/python3.9/site-packages/usb/_interop.py", line 99, in as_array
    a.fromstring(data) # deprecated since 3.2
AttributeError: 'array.array' object has no attribute 'fromstring'
```
however running `sudo python2 ./issue_cmd.py` works with python 2.7

RPwnage, 2021-03-24:
When trying to multiply two strings together. You should convert the string input from the user to a number using int() or float() in order to fix this for python3+.

stellar1s, 2021-04-26:
Issue has been patched, the entire array part has been removed. However, this only needs a simple fix for both issues.