checkra1n/PongoOS #36: Where is the code to execute in user mode after boot?

4ch12dy, 2021-01-04:
Here are two confused questions about post kernel patch.

1, I didn't found the code that setup binpack, dropbear , mount checkra1n.dmg and so on. 
how launchd process run processes like dropbear? by inject dyld?

2, any boot option/gBootFlag can skip setup loader.app or launch dropbear process after boot patched kernel?

I think the two questions maybe the same problem. Thank you so much for your response!

Siguza, 2021-01-05:
The ramdisk and binpack aren't part of Pongo. The ramdisk is currently uploaded over LLB at the same time as Pongo, and KPF merely makes the kernel boot it. That takes care of all the early boot stuff, then waits for the binpack upload over USB.

I don't understand your second question. Do you want to explicitly _not_ launch dropbear? Because we currently launch it on port 44, regardless of whether you touch loader.app...

4ch12dy, 2021-01-05:
Thanks! Sorry my bad questions, actually what I want to ask is that it can just use checkra1n-kpf-pongo to patch the kernel instead of automatically doing kind of thing like installing binpack  after boot kernel. Or can I do it manually? That mean I can run my code like upload binpack and launch dropbear.

btw, I build the project and I can load kext module, build/checkra1n-kpf-pongo, to patch kernel in pangos shell and send kpf and bootux commands to boot. But I don't know how to use build/Pongo(Mach-O 64-bit preload executable arm64) and build/Pongo.bin(maybe upload to device).  Thanks again!

Siguza, 2021-01-05:
We currently have no "nice" infrastructure in place for a custom ramdisk, but I guess you could set one with your own Pongo module...

And the way you use Pongo.bin is by passing it to `checkra1n -k`. You can also pass PongoConsolidated.bin, which includes the KPF and will automatically run it.

closed 2021-01-05 by 4ch12dy

4ch12dy, 2021-01-07:
I am not sure if it's right that checkra1n insert a ramdisk into the head of  memory-map in devicetree, which is md0.
and add "rootdev=md0" to bootargs to enable kernel load the ramdisk(mount?), then files like "/payload" will be released. This file will replace launchd and become the first launching process.
According to the above, it will do nothing if removing "rootdev=md0" bootargs, right? Also, I can add a custom ramdisk to make it.