checkra1n/PongoOS #38: Uploading files larger than 16 MB (0x1000000) fails

alula, 2021-01-12:
pongoOS version: `2.4.5-f337e98a`
Device: iPhone 7 (9,3/D101)

For some reason, any attempt to upload a file that's bigger than 16 MB fails (for example, with bundled upload_data.py script)
Seems to only happen on builds bundled with checkra1n 0.12 beta and future versions.

```
Traceback (most recent call last):
  File "../pongoOS/scripts/upload_data.py", line 34, in <module>
    dev.write(2,data,1000000)
  File "/home/alula/.pyenv/versions/3.8.3/lib/python3.8/site-packages/usb/core.py", line 943, in write
    return fn(
  File "/home/alula/.pyenv/versions/3.8.3/lib/python3.8/site-packages/usb/backend/libusb1.py", line 819, in bulk_write
    return self.__write(self.lib.libusb_bulk_transfer,
  File "/home/alula/.pyenv/versions/3.8.3/lib/python3.8/site-packages/usb/backend/libusb1.py", line 920, in __write
    _check(retval)
  File "/home/alula/.pyenv/versions/3.8.3/lib/python3.8/site-packages/usb/backend/libusb1.py", line 595, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBError: [Errno 5] Input/Output Error
```

Rabih-A, 2021-01-13:
thx friend

Rabih-A, 2021-01-13:
thx friend