checkra1n/PongoOS #39: [Question][blackbird] Is blackbird supported on iOS 12.4?

krisanovdev, 2021-01-15:
Hi, I have some problems with blackbird exploit on A10 device (7 Plus iOS 12.4).
Running checkra1n with command:
`checkra1n -c -k build/PongoConsolidated.bin -p`
Running Pongo-shell commands:
`
sep tz0
sep fwload
`
Output:
`
0x120094> 0: ff 00 ff 00 00 00 00 00 8e e4 28 6f 21 15 7c 3e 
0x120094300x120094300: 93 41 64 f7 ab a1 d9 0d 9d a7 60 87 eb 0d dc fb
`
And SEP panic on the device side.

krisanovdev, 2021-01-15:
I guess checkra1n is not disabling TZ0 lock in iBoot for iOS < 14?

Siguza, 2021-01-16:
I actually don't know if our TZ0 patch works that far back, but there's a bigger issue even: we currently rely on sep-fw.img4 being loaded into memory by iBoot, and we only boot it if `sepfw-booted` is set in the DeviceTree.  
We plan to extend this at some point; probably as a USB interface at first, and later as a NVMe+APFS driver.

827Dream, 2021-08-04:
hi ,  have you resolve it ?