checkra1n/PongoOS #40: Is it possible to boot already not signed IPSW without saved shsh in iPhone (SE) with pongoOS?

likemusic, 2021-01-17:
I need in result to backup files from iPhone SE in recovery loop boot (seems because there is not enough free storage space to upgrade to new signed ipsw - error 14 when 75% of file system sent).

Therefore I want to restore 13.x version (because it has weight less about 1 GB).

Or may be is there a way to decrease size of current signed IPSW (iOS 14.3 (18C66) - https://ipsw.me/download/iPhone8,4/18C66) and boot it, to have access to iPhone file system?

Alkhir, 2021-01-18:
[https://github.com/checkra1n/pongoOS/issues/40#issue-787672378](url)

likemusic, 2021-01-20:
#10 - related issue with some useful comments.

likemusic, 2021-01-20:
Perhaps (according to [comment](https://github.com/checkra1n/pongoOS/issues/29#issuecomment-724326872)) there will be a module that provides access to the file system.

likemusic, 2021-01-20:
#9  - seems related

nawaf965456, 2021-02-01:
14.3