checkra1n/PongoOS #50: pongoOS 1.2.1-3a324c92

tolikes, 2021-02-03:
were pongoOS 1.2.1-3a324c92 ??
I need for lunch linux arm64 on ip7

Siguza, 2021-02-03:
Why do you need that version?  
Why can't you use the version bundled with checkra1n?

tolikes, 2021-02-03:
I have tried various versions of checkra1n but it only works pongoOS OS 1.2.1
In the most modern versions they are limited to the Linux boot, I used the latest version of pongoOS i put but it does not work correctly.
Error. /load-linux.mac Linux.lzma dtbpack
Failed to transfer image to device

tolikes, 2021-02-03:
I need to do an experiment with linux, I beg you to give me access to the old files that correspond to that version

Siguza, 2021-02-03:
Hmm, that should've been fixed... have you tried with the latest build artifact from [here](https://github.com/checkra1n/pongoOS/suites/1946806974/artifacts/38506559)?

But I guess if that doesn't work, then... here you go: [Pongo-1.2.1.zip](https://github.com/checkra1n/pongoOS/files/5919412/Pongo-1.2.1.zip)

tolikes, 2021-02-03:
Thanks you finally booted Linux

closed 2021-02-03 by tolikes

funcoleto, 2021-08-08:
> Thanks you finally booted Linux

please, Could you pass me pongo.bin? because PongoConsolidated.bin does not work for me...

funcoleto, 2021-08-08:
I need pongo.bin 1.2.1, because in repository I do not see, I found 1.3 but doesn't work because for the big files does not work..

this is pongo.bin 1.3
https://mega.nz/file/rQgA3ZoL#dgPEowdZ0YLGKFywphgbmQLM1HwQ8kveuz82HNku6yM
if you want to load ./load-linux Android dtbpack on iPhone 7 work but for linux does not work...

tolikes, 2025-07-19:
> [@Shadow5704](https://github.com/Shadow5704) you might want to look at https://github.com/HoolockLinux/ if you're interested in running Linux on iDevices. The original Sandcastle project has long since abandoned, and on the working branch we removed the old Linux interface in favour of loading m1n1 from Asahi.

Nice! Time to test witch ip5s and ip7plus

tolikes, 2025-07-19:
> I tried building and got lazy to build and configure the kernel tree, so would be great if someone shared pre-compiled payloads to run on iphone 7 with debian based linux distro with USB Ethernet from computer and wifi and bluetooth working with ssh shell working(and must display ssh username and password on screen)

+1 🥲