checkra1n/PongoOS #53: Error with the load_module.py

jamesjing106, 2021-02-19:
I compiled the testmodule under example folder. Then copied it to scripts folder.
I ran the following command but met with an error. Could you help? Thanks
python3 module_load.py test_module 
Traceback (most recent call last):
  File "module_load.py", line 36, in <module>
    dev.write(2,"")
  File "/Library/Python/3.8/site-packages/usb/core.py", line 943, in write
    return fn(
  File "/Library/Python/3.8/site-packages/usb/backend/libusb1.py", line 819, in bulk_write
    return self.__write(self.lib.libusb_bulk_transfer,
  File "/Library/Python/3.8/site-packages/usb/backend/libusb1.py", line 920, in __write
    _check(retval)
  File "/Library/Python/3.8/site-packages/usb/backend/libusb1.py", line 595, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBError: [Errno 60] Operation timed out

RPwnage, 2021-03-24:
> Operation timed out 

This error seems USB Device related. Are you sure your USB Device is still connected correctly at the point you get this Error?

jamesjing106, 2021-04-13:
Yes, thanks!