checkra1n/PongoOS #56: Building on Ubuntu 20.10 fails

k1gen, 2021-02-26:
I have system in VMware Workstation 16, was working by this instruction: https://github.com/checkra1n/pongoOS#building-on-linux.
This is what i installed: cland llvm ld64 cctools-strip. 
Here's my log: https://pastebin.com/EtnjeJYP

Sorry for my bad english, btw

tntboom5656, 2021-03-23:
I'm using a regular ubuntu 20.10 computer, and that doesn't help. We need help with this!!

Siguza, 2021-03-23:
I'll have to look into this later. For now, try `CFLAGS='-Wno-error' make clean all`.

tntboom5656, 2021-03-23:
ok

tntboom5656, 2021-03-23:
that didn't help

matteyeux, 2021-04-15:
Can you try this script to build PongoOS : https://gist.github.com/matteyeux/8f286b32334a62d3f65018e5a9785caf.
I managed to build PongoOS on Ubuntu 20.10 with this script.

Siguza, 2021-04-15:
@matteyeux that simply uses an old clang version, which... I guess works, but isn't exactly proper. :/

tntboom5656, 2021-04-15:
Yeah, I gotta try it first, before I debunk it completely

Tsumiicode, 2021-04-26:
I Have the error aswell and with both even using the script it errors