checkra1n/PongoOS #58: ld: symbol(s) not found for architecture arm64

cragy0516, 2021-03-11:
....
ld: warning: ignoring file newlib/aarch64-none-darwin/lib/libc.a, building for free standing-arm64 but attempting to link with file built for unknown-unsupported file format ( 0x21 0x3C 0x61 0x72 0x63 0x68 0x3E 0x0A 0x2F 0x20 0x20 0x20 0x20 0x20 0x20 0x20 )
ld: warning: ignoring file newlib/aarch64-none-darwin/lib/libm.a, building for free standing-arm64 but attempting to link with file built for unknown-unsupported file format ( 0x21 0x3C 0x61 0x72 0x63 0x68 0x3E 0x0A 0x2F 0x20 0x20 0x20 0x20 0x20 0x20 0x20 )
Undefined symbols for architecture arm64:
  "_realloc", referenced from:
      _public_api in cc-4117cf.o
     (maybe you meant: _reallocate_loader_xfer_data)
  "_strrchr", referenced from:
      _fdt_subnode_offset_namelen in cc-4117cf.o
  "_putchar", referenced from:
      _command_main in cc-4117cf.o
      _public_api in cc-4117cf.o
  "_fgets", referenced from:
      _command_main in cc-4117cf.o
  "_strtoul", referenced from:
      _poke_cmd in cc-4117cf.o
      _sep_pwned_peek in cc-4117cf.o
      _sep_pwned_poke in cc-4117cf.o
      _sep_pwned_jump in cc-4117cf.o
      _public_api in cc-4117cf.o
  "_qsort", referenced from:
      _command_register in cc-4117cf.o
      _start_host_shell in cc-4117cf.o
  "_strtoull", referenced from:
      _md8_cmd in cc-4117cf.o
      _phys_page_dump in cc-4117cf.o
      _peek_cmd in cc-4117cf.o
      _poke_cmd in cc-4117cf.o
      _spawn_cmd in cc-4117cf.o
      _tz0_set in cc-4117cf.o
      _xnu_pf_get_kext_header in cc-4117cf.o
      ...
  "_printf", referenced from:
      _pongo_boot_linux in cc-4117cf.o
  "_strdup", referenced from:
      _modload_cmd in cc-4117cf.o
      _hal_service_op in cc-4117cf.o
      _hal_load_dtree_child_node in cc-4117cf.o
  "_vfiprintf", referenced from:
      _task_crash_internal in cc-4117cf.o
      _public_api in cc-4117cf.o
  "_strlcpy", referenced from:
      _task_list in cc-4117cf.o
  "_viprintf", referenced from:
      _panic in cc-4117cf.o
  "_strncmp", referenced from:
      _device_clock_by_name in cc-4117cf.o
      _dt_cbp in cc-4117cf.o
  "_fflush", referenced from:
      _print_state in cc-4117cf.o
      _queue_rx_char in cc-4117cf.o
      _command_main in cc-4117cf.o
  "_strlen", referenced from:
      _dt_find_cb in cc-4117cf.o
      _pongo_entry_cached in cc-4117cf.o
      _memstr in cc-4117cf.o
      _memstr_partial in cc-4117cf.o
      _command_main in cc-4117cf.o
      _linux_cmdline_cmd in cc-4117cf.o
      _aes_cmd in cc-4117cf.o
      ...
     (maybe you meant: _strlen_)
  "_fiprintf", referenced from:
      _print_state in cc-4117cf.o
      _wdt_reset in cc-4117cf.o
      _task_crash_internal in cc-4117cf.o
      _sep_handle_msg_from_sep in cc-4117cf.o
      _sep_boot_auto in cc-4117cf.o
      _seprom_fwload_race in cc-4117cf.o
      _public_api in cc-4117cf.o
[... 150 lines cut ...]
      _xnu_pf_get_kext_header in cc-4117cf.o
  "_memcmp", referenced from:
      _pongo_entry_cached in cc-4117cf.o
      _seprom_fwload_race in cc-4117cf.o
      _xnu_pf_ptr_to_data_match in cc-4117cf.o
      _fdt_get_property_namelen_ in cc-4117cf.o
      _fdt_subnode_offset_namelen in cc-4117cf.o
      _fdt_add_property_ in cc-4117cf.o
      _public_api in cc-4117cf.o
      ...
  "_strncpy", referenced from:
      _pongo_entry_cached in cc-4117cf.o
      _proc_create_task in cc-4117cf.o
      _task_create_extended in cc-4117cf.o
      _task_create in cc-4117cf.o
      _spawn_cmd in cc-4117cf.o
  "_siprintf", referenced from:
      _pongo_entry_cached in cc-4117cf.o
ld: symbol(s) not found for architecture arm64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make: *** [build/Pongo] Error 1

macOS Catalina, version 10.15.7
I reinstalled xcode command-line tools and updated xcode, clang, etc.
what should I do for it? thank you.

Siguza, 2021-03-11:
Hm, I've seen two issues cause this before:

1. clang and ld versions being out of sync.
2. GNU ar, libtool and ranlib overriding Darwin/LLVM ar, libtool and ranlib.

The first shouldn't be an issue per se if you're using Xcode, unless you transferred the project folder from a machine with newer Xcode...? In that case, a `make distclean` should do the trick.  
Otherwise, could you check if any GNU tools override their Darwin/LLVM counterparts?

cragy0516, 2021-03-12:
I checked clang and ld versions and it is not problem, maybe.
So I tried to check if any GNU tools override their Darwin/LLVM things but with no luck, so I simply changed my account to other user's and it compiled successfully.
I don't know still what happend but... it worked :-S

fahad12346, 2021-03-12:
Checkra1n

Siguza, 2021-03-19:
I'm glad you got it to work.  
But hm, I suppose it might be worth to have the Makefile check for this...