checkra1n/PongoOS #70: libc6 requirements too high for buster in checkra1n repository

asdfugil, 2021-04-28:
```
nick@nick-pc (nickchan.ddns.net):~$ sudo apt install ld64 cctools-strip
[sudo] password for nick: 
正在讀取套件清單... 完成
正在重建相依關係          
正在讀取狀態資料... 完成
有些套件無法安裝。這可能意謂著您的要求難以解決，或是若您使用的是
unstable 發行版，可能有些必要的套件尚未建立，或是被移出 Incoming 了。
以下的資訊或許有助於解決當前的情況：

下列的套件有未滿足的相依關係：
 cctools-strip : 相依關係: libc6 (>= 2.29) 但 2.28-10 卻將被安裝
 ld64 : 相依關係: libc6 (>= 2.29) 但 2.28-10 卻將被安裝
E: 無法修正問題，您保留 (hold) 了損毀的套件。
```

Siguza, 2021-04-28:
We know, but there is no easy fix.  
Compiling for an older lib from a new system is extremely hard (see [this](https://stackoverflow.com/q/2856438/2302862)) and compiling on an old system is not possible because these tools need to link against new versions of the LLVM codebase, which cannot be installed on these old systems. :|

asdfugil, 2021-04-28:
all right i'll just use bullseye

closed 2021-04-28 by asdfugil