checkra1n/PongoOS #71: 14.5 breaks blackbird

buiniyslavik, 2021-04-30:
TZ0 ends up being locked before pongoOS can do anything about it. I figure that's because of the new iBoot version.
Here's from 14.4.2:
```
#================== 
# 
# pongoOS 2.5.0-0cb6126f 
# 
# https://checkra.in 
# 
#================== 
Booted by: iBoot-6723.80.19 
Built with: Clang 12.0.0 (clang-1200.0.32.29) 
Running on: Apple A10 (T8010) 
 pongoOS> sep pwn  
image len 14b7a0 -> 14b7c8 
found victim block @ 14b7a0 
AP->SEP: endpoint ff, tag: 0, opcode: 1, param: 0, data: 0 
SEP->AP: endpoint ff, tag: 0, opcode: 65, param: 0, data: 0 
AP->SEP: endpoint ff, tag: 0, opcode: 5, param: 0, data: 0 
SEP->AP: endpoint ff, tag: 0, opcode: 69, param: 0, data: 0 
SEP->AP: endpoint ff, tag: 0, opcode: d2, param: 0, data: 2 
AP->SEP: endpoint ff, tag: 0, opcode: 1, param: 0, data: 0 
SEP->AP: endpoint ff, tag: 0, opcode: 65, param: 0, data: 0 
successfully obtained SEPROM code execution 
sepb @ 14b6c0 
patched out bpr check 
SEP payload ready to boot
```
All good. But if you restore to 14.5, you'll get this:
```
#================== 
# 
# pongoOS 2.5.0-0cb6126f 
# 
# https://checkra.in 
# 
#================== 
Booted by: iBoot-6723.102.4 
Built with: Clang 12.0.0 (clang-1200.0.32.29) 
Running on: Apple A10 (T8010) 
 pongoOS> sep pwn  
image len 152bf0 -> 152c18 
found victim block @ 152c00 
Registers are locked
```

Siguza, 2021-05-01:
Are you using checkra1n 0.12.3?

buiniyslavik, 2021-05-04:
This indeed was a checkra1n issue, thanks. Closing.

closed 2021-05-04 by buiniyslavik