checkra1n/PongoOS #72: checkra1n doesn't work but pongoOS does

mbesemann, 2021-05-05:
checkra1n ran normally gives me usbmux error -79 (Linux) or error -20 (Mac).

However, building pongoOS and specifying it with

`checkra1n -k Pongo.bin`

brings me to a PongoOS shell.

I have 2 questions:

1) How do I load the checkra1n module to continue the exploit and get a jailbreak?
2) If that fails, can I set a nonce in nvram to restore to 14.3 with pongoterm?

Siguza, 2021-05-05:
Just to check - specifiying `PongoConsolidated.bin` doesn't work?

NHLGames, 2021-05-05:
Nope, tried that - that differs from the same issue as running checkra1n normally.  I've even tried using eclipsa to go into Pwned DFU mode but it seems the device just wants to jump out of it and boot normally

Siguza, 2021-05-06:
Okay, so:

1. Extract the ramdisk from the macOS binary (`__CONST.__rdsk` segment, or use this one: [rdsk.dmg.gz](https://github.com/checkra1n/pongoOS/files/6431346/rdsk.dmg.gz))
2. Have both the ramdisk and `checkra1n-kpf-pongo` in an accessible path.
3. Get `pongoterm` from latest git master.
4. Create a command file (I'll call it `cmd.txt`) with the following contents:
    ```
    sep auto
    /send path/to/checkra1n-kpf-pongo
    modload
    /send path/to/rdsk.dmg
    ramdisk
    xargs rootdev=md0
    bootx
    ```
5. Run `pongoterm <cmd.txt` and let it wait
6. Run checkra1n.

mbesemann, 2021-05-06:
This is what I get:

```
#==================
#
# pongoOS 2.5.0-0cb6126f
#
# https://checkra.in
#
#==================
Booted by: iBoot-6723.102.4
Built with: Clang 12.0.5 (clang-1205.0.22.9)
Running on: Apple A8X (T7001)
pongoOS> Bad command: /send
pongoOS> [modload_macho:i] Attempting to load a module
[modload_macho:!] load module: short read
pongoOS> Bad command: /send
pongoOS> please upload a ramdisk before issuing this command
pongoOS> set xnu boot arg cmdline to: [rootdev=md0]
pongoOS> %
```
For reference, my cmd.txt looks like this, and the 2 files are in the same directory (I tried both with ./ and without):

```
sep auto
/send ./checkra1n-kpf-pongo
modload
/send ./rdsk.dmg.gz
ramdisk
xargs rootdev=md0
bootx
```

mbesemann, 2021-05-06:
Ok this deserves its own post because I finally did it :)

1. Run checkra1n with the following args:
checkra1n -csvk Pongo.bin

2. Only once at the PongoOS shell, run pongoterm with the aforementioned cmd.txt

3. Success - checkra1n icon is on the home screen :)

Log of run:

```
 - [05/06/21 10:05:41] <Info>: Waiting for DFU devices
 - [05/06/21 10:05:41] <Verbose>: DFU mode device found
 - [05/06/21 10:05:41] <Info>: Exploiting
 - [05/06/21 10:05:41] <Verbose>: Attempting to perform checkm8 on 7001 1...
 - [05/06/21 10:05:41] <Info>: Checking if device is ready
 - [05/06/21 10:05:41] <Verbose>: == Checkm8 Preparation stage ==
 - [05/06/21 10:05:41] <Verbose>: DFU mode device found
 - [05/06/21 10:05:41] <Info>: Setting up the exploit (this is the heap spray)
 - [05/06/21 10:05:41] <Verbose>: == Checkm8 Setup stage ==
 - [05/06/21 10:05:41] <Info>: Right before trigger (this is the real bug setup)
 - [05/06/21 10:05:41] <Verbose>: Entered initial checkm8 state after 3 steps, issuing DFU abort..
 - [05/06/21 10:05:42] <Verbose>: DFU device disconnected
 - [05/06/21 10:05:42] <Verbose>: DFU mode device found
 - [05/06/21 10:05:42] <Verbose>: == Checkm8 Trigger stage ==
 - [05/06/21 10:05:42] <Verbose>: Checkmate!
 - [05/06/21 10:05:42] <Verbose>: DFU device disconnected
 - [05/06/21 10:05:42] <Verbose>: DFU mode device found
 - [05/06/21 10:05:42] <Verbose>: == Checkm8 Trying to run payload... ==
 - [05/06/21 10:05:42] <Verbose>: If everything went correctly, you should now have code execution.
 - [05/06/21 10:05:42] <Verbose>: DFU device disconnected
 - [05/06/21 10:05:43] <Info>: Entered download mode
 - [05/06/21 10:05:43] <Verbose>: Download mode device found
 - [05/06/21 10:05:43] <Info>: Booting...
 - [05/06/21 10:05:43] <Verbose>: Setting bootargs to: rootdev=md0
 - [05/06/21 10:05:44] <Verbose>: Download mode device disconnected
 - [05/06/21 10:06:13] <Info>: All Done
 - [05/06/21 10:06:13] <Verbose>: Bootstrap already installed, done
 ```

**UPDATE**: Bad news - Installing Cydia worked, then I realized I actually wanted to install oddysseyra1n, so I restored the system from the checkra1n app, and now I can't even get to the pongoOS shell.  I'll keep trying I guess!

4ch12dy, 2021-05-08:
> Okay, so:
>
> 1. Extract the ramdisk from the macOS binary (`__CONST.__rdsk` segment, or use this one: [rdsk.dmg.gz](https://github.com/checkra1n/pongoOS/files/6431346/rdsk.dmg.gz))
> 2. Have both the [...]

rinsuki, 2021-05-15:
I have same issue, and https://github.com/checkra1n/pongoOS/issues/72#issuecomment-833177765 workaround is worked. Thanks!

in my case:
- simply run checkra1n, then I got -20 (timeout) error. I also got similar error with `-k PongoConsolidated.bin`
- attached rdsk.dmg.gz will not work (even ungzipped) in my environment, then I extracted from checkra1n 0.12.4 with this tool https://gist.github.com/C0deH4cker/80b53de22012146ea9d8
- `pongocmd` sometimes freezes until disconnect and reconnect Lightning cable

edwin170, 2023-05-27:
> Okay, so:
>
> 1. Extract the ramdisk from the macOS binary (`__CONST.__rdsk` segment, or use this one: [rdsk.dmg.gz](https://github.com/checkra1n/pongoOS/files/6431346/rdsk.dmg.gz))
> 2. Have both the [...]

that rdsk doesn't have the binpack files. may you please tell where we could find them.

Siguza, 2023-05-28:
@edwin170 `__CONST.__overlay`

dualra1n, 2023-05-30:
> @edwin170 `__CONST.__overlay`

idk if it is a dmg file but i tried it and the dmg was corrupted so i used file command "overlay.dmg: zlib compressed data" so i extracted it then it was just data. so i think that was extracted bad, however i did good i use otool to know the offset and the size, well could you share me the file as a zip or dmg or something similar ?

edwin170, 2023-06-03:
> @edwin170 `sudo hdik overlay.dmg`

sir may you please say who is responsible for mounting the overlay.dmg image?

Siguza, 2023-06-03:
@edwin170 The `payload` binary.

edwin170, 2023-06-03:
|

> @edwin170 The `payload` binary.

"private/var/tmp/checkra1n.dmg" that should be the overlay

Halconmilenario21, 2023-12-08:
Hello! To make things more difficult, I am trying to jailbreak an iPhone 7 with iOS 15.7.3 using a Raspberry Pi 4. I use CLI bulldog 0.1337.2. My device boots into PongoOS, loads KPF, ram disk and boot-args, but it doesn’t load the overlay. I always have to stop the checkra1n process from the terminal so it never says the “All Done”. I’ve decompressed the overlay but I don’t know if it’s a dmg or not. The command I use is:
./checkra1n-linux-armel -v -V -r rdsk.dmg -o overlay. Anybody has an idea?