checkra1n/PongoOS #75: Noobie problems...

doctorcloudbase, 2021-05-24:
Hi All,

I am trying to use either UnlockGo or 4UKey to get access to my old iPad Pro 9.7inch at iPadOS 14.5.1. 

Years ago I was locked out by an issue with 2-step verification and that AppleID - I had to start over with a new one. I have been 'waiting' for the software ecosystem to bypass this all and be able to essentially start over with is as a device for my trainee Doctor gf. 

I first have to remove the passcode lock, which those softwares do, but when I try to action any further steps, their is 
an 'Activation lock' (Weird, since it was long-since activated), which to bypass invokes Checkra1n. The previous version of this 0.12.3 reported that it did not support iPadOS 14.5, so again I've been waiting until Checkra1n caught up and 0.12.4 seems to be okay with 14.5, so I try running it, but once in DFU mode, it looks like PongoOS loads - it get's to a command-line prompt >PongoOS ...and then times out, with an error -20. the suggestion is that this iPad has a non-standard tweak, but 'none that I know if' unless it's something that has been introduced by UnlockGo or 4Ukey, but this would seem unlikely, since they are 'attached' and in-train..?

So, is there some additional Boot argument I can make? Verbose boot doesn't seem to result in any further info on either iPad or Checkra1n (on MacOS). 

I basically am stuck in this loop, and unless I can get this jailbroken, then there would seem to be no way around this iPad lockout. I had essentially given up on it a couple of years ago so this is more 'fun' than essential, but for all I know I am effectively 'bug reporting' on Checkra1n here, so thought I'd mentioning!

Thanks in advance.

doctorcloudbase, 2021-05-24:
PS - not sure what a 'normal' output on the iPad would be, but the last few lines are:

Found old-style rdsk!
Detected corrupted kerninfo!
Enabling USB
Done!
pongoOS>

To me, this reads as the jailbreak fooling the ipad to accepting 'USB control' by the detection of the old/corrupted issue on boot, ie 'all is well'... However, if these are errors being detected by Checkra1n itself and are resulting in the timeout, they might give some insight to anyone who knows what the go here is...

D

Siguza, 2021-05-24:
https://github.com/checkra1n/BugTracker/issues/2074

closed 2021-05-24 by Siguza