checkra1n/PongoOS #79: How to compile fdt module from `pongoOS/src/modules/linux/libfdt` ?

cuj, 2021-06-06:
Hi, I was trying to run linux using [this build](https://assets.checkra.in/downloads/sandcastle/0175ae56bcba314268d786d1239535bca245a7b126d62a767e12de48fd20f470/linux-sandcastle.zip), however, I see that `fdt` is executed in pongoOS and does nothing in my iPhone 7 Plus, so I would like to compile the fdt module from `pongoOS/src/modules/linux/libfdt` but I can't seem to make it, running `make` in that directory `pongoOS/src/modules/linux/libfdt` through terminal returns `make: *** No targets specified and no makefile found.  Stop.` even when there's a Makefile.libfdt so can you please tell me how to make fdt module from `pongoOS/src/modules/linux/libfdt`? thanks.

funcoleto, 2021-08-08:
hi, I'm try to load the same project but always stop, I arribe to PONGO OS and then with ./pongoterm I send several command like =>

/send dtbpack
fdt
/send Linux.lzma
bootl 

but then stop

did you find the solution?

cuj, 2022-05-31:
> hi, I'm try to load the same project but always stop, I arribe to PONGO OS and then with ./pongoterm I send several command like =>
> 
> /send dtbpack fdt /send Linux.lzma bootl
> 
> but then stop
> 
> did you find the solution?

Sorry, no I haven't found a solution.

closed 2022-05-31 by cuj