checkra1n/PongoOS #83: Flashless boot

fxsheep, 2021-07-02:
pongoOS normally boots as an iBoot payload, and the iBoot is loaded by SecureROM from the NAND. Can it boot from iBoot loaded in RAM(e.g. by DFU)? i.e. boot pongoOS without a NAND flash

Siguza, 2021-07-03:
The first step required for this would be raw image boot in checkra1n - which should not be a problem in theory.  
Then, if booted from DFU, then iBoot will act as iBSS/iBEC and expect a devicetree/kernel/ramdisk combo uploaded over USB. And our injected shellcode would divert the boot flow to PongoOS after the handoff trampoline. That should in theory all work, if we implemented raw image boot.

However, if your device does physically not have a functional NAND, then that will be an issue, as iBoot will try to load NVRAM variables and SysCfg data off NAND.

DesentivsedToGore, 2022-04-16:
> The first step required for this would be raw image boot in checkra1n - which should not be a problem in theory. Then, if booted from DFU, then iBoot will act as iBSS/iBEC and expect a [...]

could you add it as a option instead? i do see the light in this suggestion.

fxsheep, 2023-10-21:
iPhone7,2 boots up to SSH ramdisk properly with NAND physically desoldered. (Yes, it does complain about absense of NAND)

edwin170, 2023-11-28:
> The first step required for this would be raw image boot in checkra1n - which should not be a problem in theory. Then, if booted from DFU, then iBoot will act as iBSS/iBEC and expect a [...]

and if we just patched the iboot to localboot automatically and then load it on dfu it should boot and load devicetree/kernel/ramdisk from the system instead of usb, but there is the problem it will boot without checkra1n. is there something to get checkra1n inject the shellcode into an specific iboot image loaded from usb and then boot with it ?