checkra1n/PongoOS #85: Undefined symbols for architecture arm64

theIdeviceguy18383, 2021-07-17:
im trying to compile on Linux, I have all the dependencies, but i get this error during compilation 
make  -C newlib all
make[1]: Entering directory '/home/ryan/iOS-downgrade-tethered/pongoOS/newlib'
make  -C /home/ryan/iOS-downgrade-tethered/pongoOS/newlib/build all
make[2]: Entering directory '/home/ryan/iOS-downgrade-tethered/pongoOS/newlib/build'
make "AR_FLAGS=rc" "CC_FOR_BUILD=gcc" "CFLAGS=--target=arm64-apple-ios12.0 -Wall -O3 -ffreestanding -nostdlib -nostdlibinc -fno-builtin -fno-blocks -U__nonnull -D_LDBL_EQ_DBL " "CCASFLAGS=--target=arm64-apple-ios12.0 -Wall -O3 -ffreestanding -nostdlib -nostdlibinc -fno-builtin -fno-blocks -U__nonnull -D_LDBL_EQ_DBL " "CFLAGS_FOR_BUILD=" "CFLAGS_FOR_TARGET=" "INSTALL=/usr/bin/install -c" "LDFLAGS=" "LIBCFLAGS=" "LIBCFLAGS_FOR_TARGET=" "MAKE=make" "MAKEINFO=/bin/bash /home/ryan/iOS-downgrade-tethered/pongoOS/newlib/src/missing --run makeinfo " "PICFLAG=" "PICFLAG_FOR_TARGET=" "SHELL=/bin/bash" "EXPECT=`if test -f ./../expect/expect; then echo ./../expect/expect; else echo expect; fi`" "RUNTEST=`if test -f /home/ryan/iOS-downgrade-tethered/pongoOS/newlib/src/newlib/../dejagnu/runtest; then echo /home/ryan/iOS-downgrade-tethered/pongoOS/newlib/src/newlib/../dejagnu/runtest; else echo runtest; fi`" "RUNTESTFLAGS=" "exec_prefix=/home/ryan/iOS-downgrade-tethered/pongoOS/newlib" "infodir=/home/ryan/iOS-downgrade-tethered/pongoOS/newlib/share/info" "libdir=/home/ryan/iOS-downgrade-tethered/pongoOS/newlib/lib" "prefix=/home/ryan/iOS-downgrade-tethered/pongoOS/newlib" "tooldir=/home/ryan/iOS-downgrade-tethered/pongoOS/newlib/aarch64-none-darwin" "top_toollibdir=/home/ryan/iOS-downgrade-tethered/pongoOS/newlib/aarch64-none-darwin/lib" "AR=llvm-ar" "AS=as" "CC=clang -I/home/ryan/iOS-downgrade-tethered/pongoOS/newlib/build/targ-include -I/home/ryan/iOS-downgrade-tethered/pongoOS/newlib/src/newlib/libc/include" "LD=" "LIBCFLAGS=" "NM=" "PICFLAG=" "RANLIB=llvm-ranlib" "DESTDIR=" all-recursive
make[3]: Entering directory '/home/ryan/iOS-downgrade-tethered/pongoOS/newlib/build'
true "AR_FLAGS=rc" "CC_FOR_BUILD=gcc" "CFLAGS=--target=arm64-apple-ios12.0 -Wall -O3 -ffreestanding -nostdlib -nostdlibinc -fno-builtin -fno-blocks -U__nonnull -D_LDBL_EQ_DBL " "CCASFLAGS=--target=arm64-apple-ios12.0 -Wall -O3 -ffreestanding -nostdlib -nostdlibinc -fno-builtin -fno-blocks -U__nonnull -D_LDBL_EQ_DBL " "CFLAGS_FOR_BUILD=" "CFLAGS_FOR_TARGET=" "INSTALL=/usr/bin/install -c" "LDFLAGS=" "LIBCFLAGS=" "LIBCFLAGS_FOR_TARGET=" "MAKE=make" "MAKEINFO=/bin/bash /home/ryan/iOS-downgrade-tethered/pongoOS/newlib/src/missing --run makeinfo  " "PICFLAG=" "PICFLAG_FOR_TARGET=" "SHELL=/bin/bash" "EXPECT=expect" "RUNTEST=runtest" "RUNTESTFLAGS=" "exec_prefix=/home/ryan/iOS-downgrade-tethered/pongoOS/newlib" "infodir=/home/ryan/iOS-downgrade-tethered/pongoOS/newlib/share/info" "libdir=/home/ryan/iOS-downgrade-tethered/pongoOS/newlib/lib" "prefix=/home/ryan/iOS-downgrade-tethered/pongoOS/newlib" "tooldir=/home/ryan/iOS-downgrade-tethered/pongoOS/newlib/aarch64-none-darwin" "top_toollibdir=/home/ryan/iOS-downgrade-tethered/pongoOS/newlib/aarch64-none-darwin/lib" "AR=llvm-ar" "AS=as" "CC=clang -I/home/ryan/iOS-downgrade-tethered/pongoOS/newlib/build/targ-include -I/home/ryan/iOS-downgrade-tethered/pongoOS/newlib/src/newlib/libc/include" "LD=" "LIBCFLAGS=" "NM=" "PICFLAG=" "RANLIB=llvm-ranlib" "DESTDIR=" DO=all multi-do # make
Making all in libc
make[4]: Entering directory '/home/ryan/iOS-downgrade-tethered/pongoOS/newlib/build/libc'
Making all in argz
make[5]: Entering directory '/home/ryan/iOS-downgrade-tethered/pongoOS/newlib/build/libc/argz'
make[5]: Nothing to be done for 'all'.
make[5]: Leaving directory '/home/ryan/iOS-downgrade-tethered/pongoOS/newlib/build/libc/argz'
Making all in stdlib
make[5]: Entering directory '/home/ryan/iOS-downgrade-tethered/pongoOS/newlib/build/libc/stdlib'
make[5]: Nothing to be done for 'all'.
make[5]: Leaving directory '/home/ryan/iOS-downgrade-tethered/pongoOS/newlib/build/libc/stdlib'
Making all in ctype
make[5]: Entering directory '/home/ryan/iOS-downgrade-tethered/pongoOS/newlib/build/libc/ctype'
make[5]: Nothing to be done for 'all'.
make[5]: Leaving directory '/home/ryan/iOS-downgrade-tethered/pongoOS/newlib/build/libc/ctype'
Making all in search
make[5]: Entering directory '/home/ryan/iOS-downgrade-tethered/pongoOS/newlib/build/libc/search'
make[5]: Nothing to be done for 'all'.
make[5]: Leaving directory '/home/ryan/iOS-downgrade-tethered/pongoOS/newlib/build/libc/search'
Making all in stdio
make[5]: Entering directory '/home/ryan/iOS-downgrade-tethered/pongoOS/newlib/build/libc/stdio'
make[5]: Nothing to be done for 'all'.
make[5]: Leaving directory '/home/ryan/iOS-downgrade-tethered/pongoOS/newlib/build/libc/stdio'
Making all in string
make[5]: Entering directory '/home/ryan/iOS-downgrade-tethered/pongoOS/newlib/build/libc/string'
make[5]: Nothing to be done for 'all'.
make[5]: Leaving directory '/home/ryan/iOS-downgrade-tethered/pongoOS/newlib/build/libc/string'
Making all in signal
make[5]: Entering directory '/home/ryan/iOS-downgrade-tethered/pongoOS/newlib/build/libc/signal'
make[5]: Nothing to be done for 'all'.
make[5]: Leaving directory '/home/ryan/iOS-downgrade-tethered/pongoOS/newlib/build/libc/signal'
Making all in time
make[5]: Entering directory '/home/ryan/iOS-downgrade-tethered/pongoOS/newlib/build/libc/time'
make[5]: Nothing to be done for 'all'.
make[5]: Leaving directory '/home/ryan/iOS-downgrade-tethered/pongoOS/newlib/build/libc/time'
Making all in locale
make[5]: Entering directory '/home/ryan/iOS-downgrade-tethered/pongoOS/newlib/build/libc/locale'
make[5]: Nothing to be done for 'all'.
make[5]: Leaving directory '/home/ryan/iOS-downgrade-tethered/pongoOS/newlib/build/libc/locale'
Making all in sys
make[5]: Entering directory '/home/ryan/iOS-downgrade-tethered/pongoOS/newlib/build/libc/sys'
Making all in .
make[6]: Entering directory '/home/ryan/iOS-downgrade-tethered/pongoOS/newlib/build/libc/sys'
make[6]: Nothing to be done for 'all-am'.
make[6]: Leaving directory '/home/ryan/iOS-downgrade-tethered/pongoOS/newlib/build/libc/sys'
make[5]: Leaving directory '/home/ryan/iOS-downgrade-tethered/pongoOS/newlib/build/libc/sys'
Making all in reent
make[5]: Entering directory '/home/ryan/iOS-downgrade-tethered/pongoOS/newlib/build/libc/reent'
make[5]: Nothing to be done for 'all'.
make[5]: Leaving directory '/home/ryan/iOS-downgrade-tethered/pongoOS/newlib/build/libc/reent'
Making all in errno
make[5]: Entering directory '/home/ryan/iOS-downgrade-tethered/pongoOS/newlib/build/libc/errno'
make[5]: Nothing to be done for 'all'.
make[5]: Leaving directory '/home/ryan/iOS-downgrade-tethered/pongoOS/newlib/build/libc/errno'
Making all in misc
make[5]: Entering directory '/home/ryan/iOS-downgrade-tethered/pongoOS/newlib/build/libc/misc'
make[5]: Nothing to be done for 'all'.
make[5]: Leaving directory '/home/ryan/iOS-downgrade-tethered/pongoOS/newlib/build/libc/misc'
Making all in machine
make[5]: Entering directory '/home/ryan/iOS-downgrade-tethered/pongoOS/newlib/build/libc/machine'
Making all in aarch64
make[6]: Entering directory '/home/ryan/iOS-downgrade-tethered/pongoOS/newlib/build/libc/machine/aarch64'
make[6]: Nothing to be done for 'all'.
[... 215 lines cut ...]
      pongo$exports in lto.o
  "_frexpl", referenced from:
      pongo$exports in lto.o
  "_sinhl", referenced from:
      __cchshl in libm.a(lib_a-cephes_subrl.o)
      _ccoshl in libm.a(lib_a-ccoshl.o)
      _csinhl in libm.a(lib_a-csinhl.o)
      _ctanhl in libm.a(lib_a-ctanhl.o)
      _ctanl in libm.a(lib_a-ctanl.o)
      pongo$exports in lto.o
  "_fmodl", referenced from:
      pongo$exports in lto.o
  "_fminl", referenced from:
      pongo$exports in lto.o
  "_fmaxl", referenced from:
      pongo$exports in lto.o
  "_rintl", referenced from:
      pongo$exports in lto.o
  "_fmal", referenced from:
      pongo$exports in lto.o
  "_floorl", referenced from:
      pongo$exports in lto.o
ld: symbol(s) not found for architecture arm64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make: *** [Makefile:99: build/Pongo] Error 1

Siguza, 2021-07-17:
That sounds weird. Could it be a holdover from a pervious "failed" compilation? Does it persist if you do ` make -C newlib distclean;  make -C newlib all`?  
If it does, could you post the output of `llvm-nm newlib/aarch64-none-darwin/lib/libm.a`?

theIdeviceguy18383, 2021-07-18:
running make -C newlib distclean; make -C newlib all worked, thanks!

closed 2021-07-18 by theIdeviceguy18383

mh-cbon, 2022-04-11:
I am having this issue too.

```
$ EMBEDDED_LDFLAGS=-fuse-ld=$(pwd)/../ld64/ld64 STRIP=$(pwd)/../ld64/cctools-strip make all
...

clang -o build/Pongo --target=arm64-apple-ios12.0 -std=gnu17 -Wall -Wunused-label -Werror -O3 -flto -ffreestanding -U__nonnull -nostdlibinc -DTARGET_OS_OSX=0 -DTARGET_OS_MACCATALYST=0 -Inewlib/aarch64-none-darwin/include -nostdlib -static -Wl,-fatal_warnings -Wl,-dead_strip -Wl,-Z -fuse-ld=/home/mh-cbon/testing/PongoOS/../ld64/ld64  -DPONGO_VERSION='"2.5.1-4c9b7541"' -DAUTOBOOT -DPONGO_PRIVATE=1 -Isrc/lib -Iinclude -Iapple-include -Iinclude/modules/linux/ -Isrc/kernel -Isrc/drivers -Isrc/modules/linux/libfdt -Lnewlib/aarch64-none-darwin/lib -lc -lm -Wl,-preload -Wl,-no_uuid -Wl,-e,start -Wl,-order_file,src/sym_order.txt -Wl,-image_base,0x100000000 -Wl,-sectalign,__DATA,__common,0x8 -Wl,-segalign,0x4000 -DDER_TAG_SIZE=8 src/boot/entry.S  src/boot/stage3.c  src/boot/clearhook.S  src/boot/patches.S  src/boot/demote_patch.S  src/boot/jump_to_image.S  src/boot/main.c src/kernel/dtree.c src/kernel/dtree_getprop.c src/kernel/entry.c src/kernel/legacy.c src/kernel/locks.c src/kernel/lowlevel.c src/kernel/main_task.c src/kernel/mm.c src/kernel/panic.c src/kernel/syscall.c src/kernel/task.c src/kernel/vfs.c src/kernel/support/exit.c src/kernel/support/io.c src/kernel/support/malloc.c src/kernel/support/process.c src/kernel/support/sbrk.c src/dynamic/modload.c src/dynamic/modload_macho.c src/kernel/int.S src/shell/autoboot.c src/shell/command.c src/shell/hex.c src/shell/linux.c src/shell/main.c src/shell/usbloader.c src/drivers/aes/aes_a7.c src/drivers/aes/aes_a9.c src/drivers/aes/aes.c src/drivers/framebuffer/fb.c src/drivers/gpio/gpio.c src/drivers/hal/hal.c src/drivers/mipi/mipi.c src/drivers/plat/s5l8960.c src/drivers/plat/s8000.c src/drivers/plat/s8001.c src/drivers/plat/s8003.c src/drivers/plat/t7000.c src/drivers/plat/t7001.c src/drivers/plat/t8010.c src/drivers/plat/t8011.c src/drivers/plat/t8012.c src/drivers/plat/t8015.c src/drivers/recfg/recfg.c src/drivers/recfg/recfg_soc.c src/drivers/sep/sep.c src/drivers/timer/timer.c src/drivers/tz/tz.c src/drivers/uart/uart.c src/drivers/usb/synopsys_otg.c src/drivers/xnu/xnu.c src/drivers/sep/sep_racer.S src/drivers/xnu/xnu.S src/modules/linux/libfdt/fdt_addresses.c src/modules/linux/libfdt/fdt.c src/modules/linux/libfdt/fdt_empty_tree.c src/modules/linux/libfdt/fdt_overlay.c src/modules/linux/libfdt/fdt_ro.c src/modules/linux/libfdt/fdt_rw.c src/modules/linux/libfdt/fdt_strerror.c src/modules/linux/libfdt/fdt_sw.c src/modules/linux/libfdt/fdt_wip.c src/modules/linux/linux.c src/lib/img4/img4.c src/lib/libDER/DER_Decode.c src/lib/libDER/DER_Encode.c src/lib/libDER/oids.c src/lib/lzma/lzmadec.c
ld: warning: ignoring file /tmp/stage3-bd29a7.o, building for free standing-arm64 but attempting to link with file built for unknown-unsupported file format ( 0xDE 0xC0 0x17 0x0B 0x00 0x00 0x00 0x00 0x14 0x00 0x00 0x00 0x7C 0x25 0x00 0x00 )
ld: warning: ignoring file /tmp/main-c9a71e.o, building for free standing-arm64 but attempting to link with file built for unknown-unsupported file format ( 0xDE 0xC0 0x17 0x0B 0x00 0x00 0x00 0x00 0x14 0x00 0x00 0x00 0x34 0x0B 0x00 0x00 )
ld: warning: ignoring file /tmp/dtree-efe2e7.o, building for free standing-arm64 but attempting to link with file built for unknown-unsupported file format ( 0xDE 0xC0 0x17 0x0B 0x00 0x00 0x00 0x00 0x14 0x00 0x00 0x00 0x70 0x17 0x00 0x00 )
ld: warning: ignoring file /tmp/dtree_getprop-052e05.o, building for free standing-arm64 but attempting to link with file built for unknown-unsupported file format ( 0xDE 0xC0 0x17 0x0B 0x00 0x00 0x00 0x00 0x14 0x00 0x00 0x00 0xFC 0x11 0x00 0x00 )
ld: warning: ignoring file /tmp/entry-921e1b.o, building for free standing-arm64 but attempting to link with file built for unknown-unsupported file format ( 0xDE 0xC0 0x17 0x0B 0x00 0x00 0x00 0x00 0x14 0x00 0x00 0x00 0xA4 0x38 0x00 0x00 )
ld: warning: ignoring file /tmp/lowlevel-e63d81.o, building for free standing-arm64 but attempting to link with file built for unknown-unsupported file format ( 0xDE 0xC0 0x17 0x0B 0x00 0x00 0x00 0x00 0x14 0x00 0x00 0x00 0x90 0x74 0x00 0x00 )ld: warning: ignoring file /tmp/locks-602084.o, building for free standing-arm64 but attempting to link with file built for unknown-unsupported file format ( 0xDE 0xC0 0x17 0x0B 0x00 0x00 0x00 0x00 0x14 0x00 0x00 0x00 0x54 0x12 0x00 0x00 )
ld: warning: ignoring file /tmp/main_task-486767.o, building for free standing-arm64 but attempting to link with file built for unknown-unsupported file format ( 0xDE 0xC0 0x17 0x0B 0x00 0x00 0x00 0x00 0x14 0x00 0x00 0x00 0x58 0x15 0x00 0x00 )
ld: warning: ignoring file /tmp/mm-c09dea.o, building for free standing-arm64 but attempting to link with file built for unknown-unsupported file format ( 0xDE 0xC0 0x17 0x0B 0x00 0x00 0x00 0x00 0x14 0x00 0x00 0x00 0xF8 0x8A 0x00 0x00 )
ld: warning: ignoring file /tmp/panic-80b9af.o, building for free standing-arm64 but attempting to link with file built for unknown-unsupported file format ( 0xDE 0xC0 0x17 0x0B 0x00 0x00 0x00 0x00 0x14 0x00 0x00 0x00 0x08 0x17 0x00 0x00 )ld: warning: ignoring file /tmp/legacy-210949.o, building for free standing-arm64 but attempting to link with file built for unknown-unsupported file format ( 0xDE 0xC0 0x17 0x0B 0x00 0x00 0x00 0x00 0x14 0x00 0x00 0x00 0x80 0x0F 0x00 0x00 )
[... cut ...]
      _task_load_asserted in int-7ee3ab.o
  "_sep_check_mailbox", referenced from:
      _sep_racer in sep_racer-960391.o
  "_sep_fast_check_mailbox", referenced from:
      _sep_racer in sep_racer-960391.o
ld: symbol(s) not found for architecture arm64
clang-13: error: linker command failed with exit code 1 (use -v to see invocation)
make: *** [Makefile:99 : build/Pongo] Erreur 1
```

the output of `llvm-nm newlib/aarch64-none-darwin/lib/libm.a` is 
[llvm.txt](https://github.com/checkra1n/PongoOS/files/8465850/llvm.txt)