checkra1n/PongoOS #86: About "sep peek [address]" stuck ?

827Dream, 2021-08-04:
device : iphone7
ios version : ios 14.6
```
Booted by: iBoot-6723.120.36
Built with: Clang 12.0.5 (clang-1205.0.22.11)
Running on: Apple A10 (T8010)
[modload_macho:i] Attempting to load a module
[modload_macho:+] Loaded module checkra1n-kpf2-12.0,14.5

#==================
#
# checkra1n kpf 0.12.4
#
# Proudly written in nano
# (c) 2019-2021 Kim Jong Cracks
#
# This software is not for sale
# If you purchased this, please
# report the seller.
#
# Get it for free at https://checkra.in
#
#====  Made by  ===
# argp, axi0mx, danyl931, jaywalker, kirb, littlelailo, nitoTV
# never_released, nullpixel, pimskeks, qwertyoruiop, sbingner, siguza
#==== Thanks to ===
# haifisch, jndok, jonseals, xerub, lilstevie, psychotea, sferrini
# Cellebrite (ih8sn0w, cjori, ronyrus et al.)
#==================
Found old-style rdsk!
Pongo shell requested, stopping here!
pongoOS> Set xnu boot arg cmdline to: [rootdev=md0]
pongoOS> sep pwn
pongoOS> tz      
TZ0 (locked):
    base: 17e09c (97e09c000)
    end:  7ed5b (87ed5c000)

TZ1 (unlocked):
    base: 0 (800000000)
    end:  0 (800001000)

pongoOS> sep peek 87ea5c000

!!stuck here!!
```
when i run sep peek at 87ea5c000 it  stuck....anyone can tell me the reason？  
https://raw.githubusercontent.com/windknown/presentations/master/Attack_Secure_Boot_of_SEP.pdf
i read the part about Bypass SEP External Memory Isolation and i think i can read the memory (eg:0x87ea5c000)after "sep pwn" because the TZ0 base is changed . Wrong i ?

Siguza, 2021-08-04:
The command takes a 32-bit address. 64-bit physical addresses are not currently supported. Also, it seems the command silently truncates addresses to 32-bit...

I guess we have to make the whole SEP code more robust.

ddyw, 2021-08-30:
i got the same problem