checkra1n/PongoOS #90: Project Sandbox PongoOS error 14.7

306bobby, 2021-08-23:
I have been working on getting project sandbox to work on iOS 14.7 and on all new versions of checkra1n/pongoOS it fails. Version 10.1 boots into android fine but touchscreen does not work, 10.2 is less consistent but will boot without touchscreen functionality. The issues get worse from there

11.0: crashes in the pongo shell
12.0-12.1: pongo does not boot
12.2-present: pongo does not allow the transfer of the image

Paddy613, 2021-08-30:
Hello-World

Paddy613, 2021-08-30:
> I have been working on getting project sandbox to work on iOS 14.7 and on all new versions of checkra1n/pongoOS it fails. Version 10.1 boots into android fine but touchscreen does not work, 10.2 is less consistent but will boot without touchscreen functionality. The issues get worse from there
> 
> 10.1-10.2: boots but touchscreen isn't functional
> 
> 11.0: crashes in the pongo shell
> 
> 12.0-12.1: pongo does not boot
> 
> 12.2-present: pongo does not allow the transfer of the image

****

closed 2021-10-31 by 306bobby