checkra1n/PongoOS #93: SEP exploit unable to run on T8012, 'Registers are locked' error

BogChanos, 2021-09-16:
I am having issues accessing SEP tools on the below versions/hardware:

`#==================`
`#`
`# pongoOS 2.5.1-217eae6f`
`#`
`# https://checkra.in`
`#`
`#==================`
`Booted by: iBoot-6723.140.2`
`Built with: Clang 12.0.5 (clang-1205.0.22.11)`
`Running on: Apple T2 (T8012)`
`[modload_macho:i] Attempting to load a module`
`[modload_macho:+] Loaded module checkra1n-kpf2-12.0,14.5`

`#==================`
`#`
`# checkra1n kpf 0.12.4`

When I try to run SEP exploits, I see

`pongoOS> sep auto `
`pongoOS> sep peek`
`sep is not pwned!`
`pongoOS> sep pwn`
`Registers are locked`
`pongoOS>`

Is anyone able to further identify the issue? My best guess is either the iBoot version has broken the blackbird exploit?
Or that there is no iBoot patch for this version (but then you'd have thought PongoOS wouldn't boot at all...) 

I have tried manually patching the iBoot but I have no idea how to actually run the patched version as it seems existing tools are all built for other chip versions and checkra1n does not have an option for custom iBoot (I'm sure that would probably break checkra1n as well).

The exact revision of my board 'iBridge2,15' is listed as 'checkm8/blackbird confirmed' so I'm not sure if the latest iBoot has broken it or I'm just doing something wrong?

Any help greatly appreciated!

nyuszika7h, 2021-09-16:
I think it only works on T8010 and T8011 right now, the rest is theoretically possible but wasn't implemented yet.

Siguza, 2021-09-16:
T8012 is missing an iBoot patch - I shared a build on Discord a while ago that had such a patch, but officially, the TrustZone unlock for devices other than t8010 and t8011 is planned for a future release of checkra1n.

BogChanos, 2021-09-20:
> T8012 is missing an iBoot patch - I shared a build on Discord a while ago that had such a patch, but officially, the TrustZone unlock for devices other than t8010 and t8011 is planned for a future release of checkra1n.

I see, thanks for the info! I guess it makes sense as other chips are more relevant to most people!

Do you have a link or channel name for that build you posted on Discord? I'd be interested in taking a look