checkra1n/PongoOS #96: Running pongoOS after iOS 15.0 update

jiska2342, 2021-10-05:
Hi,

I was trying to boot pongoOS on an iPhone 7 after updating to iOS 15.0. I'm aware that the iOS 15 branch is probably still under development and I might have to wait for an according `checkra1n` release. However, I saw that people got `xnuspy` running on iOS 15, which requires pongoOS, and gave it a try on my own. In case this isn't supposed to work yet with an official release, feel free to close the ticket.

My working setup is as follows: iPhone 8, iOS 14.8, in `main` and `iOS15` branch. If I boot from pongoOS into iOS using the `iOS15` branch, I get the message *Checkra1n has started in safe mode due to an internal error*, but everything seems to work fine.

My non-working setup is: iPhone 7, iOS 15.0, in the `iOS15` branch. I don't get anything after *Right before trigger (this is the real bug setup)* in the `checkra1n` interface, the iPhone 7 is stuck without ending up in a pongoOS shell. I can still reboot the phone with the usual button combos.

BrandonPlank, 2021-10-05:
Hey, i believe you are able to boot into pongoOS if you have a JTAG cable and manually "reset" the device state using it. I haven't personally tried it because my hackintosh decided to kick the bucket. But i do have a Kong cable that is capable of JTAG.

Siguza, 2021-10-05:
Both the author of xnuspy and Sparkey (who submitted the PR with iOS 15 support) have access to the checkra1n source, including the iOS 15 development branch.

jiska2342, 2021-10-05:
@Siguza thanks for the clarification, so I guess I'll have to wait :)

BrandonPlank, 2021-10-05:
Oh, thanks for the comment siguza!

HaziqRajputzada, 2021-11-17:
@Siguza Can you please tell us any date, when Checkra1n is releasing jailbreak for IOS 15 officially?

D0lcegabbana, 2021-11-19:
> I was trying to boot pongoOS on an iPhone 7 after updating to iOS 15.0. I'm aware that the iOS 15 branch is probably still under development and I might have to wait for an according `checkra1n` [...]

Hi, how you boot pongoOs on iOS 15?

Amachik, 2021-11-20:
> Hi, how you boot pongoOs on iOS 15?

u can not.

FakeFlasher, 2021-11-24:
Iraha Checkra1n teh Update na eeeuy....Lieur Aing mah ieu pagawean numpuk Barudak

Neelhridoy6335, 2021-11-24:
@Siguza Will checkra1n be available this month for ios 15 ?

007McCandlish, 2021-11-29:
Hello everyone, I'm kinda worried about when exactly the Checkra1n update for iOS 15.0 upward will be release?