checkra1n/PongoOS #129: ld: symbol(s) not found for architecture arm64

aalyatim, 2022-04-16:
I can't compile on ubuntu 20.04  from branch iOS15 I always get this error:
ld: symbol(s) not found for architecture arm64
I tried the other suggestions posted here but that did not work(make -C newlib distclean, make -C newlib all). It compiles fine with the master branch just not this one. 

Here is the log:

clang -o build/Pongo src/boot/entry.S  src/boot/stage3.c  src/boot/clearhook.S  src/boot/patches.S  src/boot/demote_patch.S  src/boot/jump_to_image.S  src/boot/main.c src/kernel/panic.c src/kernel/syscall.c src/kernel/entry.c src/kernel/task.c src/kernel/lowlevel.c src/kernel/dtree.c src/kernel/legacy.c src/kernel/locks.c src/kernel/dtree_getprop.c src/kernel/main_task.c src/kernel/mm.c src/kernel/vfs.c src/kernel/support/exit.c src/kernel/support/io.c src/kernel/support/process.c src/kernel/support/sbrk.c src/kernel/support/malloc.c src/dynamic/modload.c src/dynamic/modload_macho.c src/kernel/int.S src/shell/autoboot.c src/shell/hex.c src/shell/usbloader.c src/shell/main.c src/shell/linux.c src/shell/command.c src/drivers/tz/tz.c src/drivers/framebuffer/fb.c src/drivers/xnu/xnu.c src/drivers/plat/s8000.c src/drivers/plat/t7000.c src/drivers/plat/t8011.c src/drivers/plat/s8003.c src/drivers/plat/t8012.c src/drivers/plat/s5l8960.c src/drivers/plat/t8010.c src/drivers/plat/s8001.c src/drivers/plat/t7001.c src/drivers/plat/t8015.c src/drivers/usb/synopsys_otg.c src/drivers/mipi/mipi.c src/drivers/hal/hal.c src/drivers/aes/aes_a7.c src/drivers/aes/aes_a9.c src/drivers/aes/aes.c src/drivers/sep/sep.c src/drivers/timer/timer.c src/drivers/recfg/recfg.c src/drivers/recfg/recfg_soc.c src/drivers/gpio/gpio.c src/drivers/uart/uart.c src/drivers/xnu/xnu.S src/drivers/sep/sep_racer.S src/modules/linux/libfdt/fdt.c src/modules/linux/libfdt/fdt_rw.c src/modules/linux/libfdt/fdt_sw.c src/modules/linux/libfdt/fdt_empty_tree.c src/modules/linux/libfdt/fdt_overlay.c src/modules/linux/libfdt/fdt_wip.c src/modules/linux/libfdt/fdt_strerror.c src/modules/linux/libfdt/fdt_ro.c src/modules/linux/libfdt/fdt_addresses.c src/modules/linux/linux.c src/lib/lzma/lzmadec.c src/lib/libDER/DER_Decode.c src/lib/libDER/DER_Encode.c src/lib/libDER/oids.c src/lib/img4/img4.c --target=arm64-apple-ios12.0 -std=gnu17 -Wall -Wunused-label -Werror -flto -ffreestanding -U__nonnull -nostdlibinc -DTARGET_OS_OSX=0 -DTARGET_OS_MACCATALYST=0 -Inewlib/aarch64-none-darwin/include -nostdlib -static -Wl,-fatal_warnings -Wl,-dead_strip -Wl,-Z -fuse-ld='/usr/bin/ld64'  -Os -moutline -DPONGO_VERSION='"2.5.1-0f9f5a87"' -DPONGO_BUILD='"0f9f5a8718c4be0f30526ed30bca05aa9c9d0efe (iOS15, dirty)"' -DAUTOBOOT -DPONGO_PRIVATE=1 -Isrc/lib -Iinclude -Iapple-include -Iinclude/modules/linux/ -Isrc/kernel -Isrc/drivers -Isrc/modules/linux/libfdt -Lnewlib/aarch64-none-darwin/fixup -lc -Wl,-preload -Wl,-no_uuid -Wl,-e,start -Wl,-order_file,src/sym_order.txt -Wl,-image_base,0x100000000 -Wl,-sectalign,__DATA,__common,0x8 -Wl,-segalign,0x4000 -DDER_TAG_SIZE=8
Undefined symbols for architecture arm64:
  "___muloti4", referenced from:
      _reallocarray in libc.a(lib_a-reallocarray.o)
ld: symbol(s) not found for architecture arm64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make: *** [Makefile:143: build/Pongo] Error 1

Siguza, 2022-04-17:
Yeah we need to update the toolchain instructions. You need compiler-rt for iOS, which is only shipped in the Darwin build of clang.

Assuming you have the standard clang of Ubuntu 20.04 (clang-10), then you can download the corresponding Darwin build from [here](https://github.com/llvm/llvm-project/releases/download/llvmorg-10.0.0/clang+llvm-10.0.0-x86_64-apple-darwin.tar.xz) (other versions can be found [here](https://github.com/llvm/llvm-project/releases)) and add the compiler-rt libraries to your toolchain like so:

    sudo tar -xf clang+llvm-10.0.0-x86_64-apple-darwin.tar.xz -C /usr/lib/llvm-10/lib/clang/10.0.0/lib --strip-components 5 '*/lib/clang/*/lib/darwin'

aalyatim, 2022-04-18:
Its searching in the include folder instead of the lib folder, is that normal:

ignoring nonexistent directory "include/modules/linux/"
#include "..." search starts here:
#include <...> search starts here:
 newlib/aarch64-none-darwin/include
 src/lib
 include
 apple-include
 src/kernel
 src/drivers
 src/modules/linux/libfdt
 /usr/lib/llvm-10/lib/clang/10.0.0/include
End of search list.
 "/usr/bin/ld64" -static -arch arm64 -iphoneos_version_min 12.0.0 -mllvm -enable-machine-outliner -mllvm -enable-linkonceodr-outlining -o build/Pongo -Lnewlib/aarch64-none-darwin/fixup /tmp/entry-0538c0.o /tmp/stage3-85455b.o /tmp/clearhook-1ffc70.o /tmp/patches-2c254f.o /tmp/demote_patch-5c1c01.o /tmp/jump_to_image-70ce69.o /tmp/main-b85dda.o /tmp/panic-c361e8.o /tmp/syscall-b35c05.o /tmp/entry-d77de6.o /tmp/task-55613c.o /tmp/lowlevel-ae40c2.o /tmp/dtree-a85093.o /tmp/legacy-947e17.o /tmp/locks-3eebcd.o /tmp/dtree_getprop-112726.o /tmp/main_task-4d48d0.o /tmp/mm-a79072.o /tmp/vfs-307ae8.o /tmp/exit-1260d2.o /tmp/io-de3f56.o /tmp/process-593a20.o /tmp/sbrk-ac73ce.o /tmp/malloc-60fdad.o /tmp/modload-5cfbcc.o /tmp/modload_macho-e9a180.o /tmp/int-a28164.o /tmp/autoboot-0c4f9f.o /tmp/hex-cebc97.o /tmp/usbloader-870398.o /tmp/main-306cbe.o /tmp/linux-851077.o /tmp/command-47396c.o /tmp/tz-82a3e3.o /tmp/fb-a7baa4.o /tmp/xnu-2d499f.o /tmp/s8000-71581c.o /tmp/t7000-f632f9.o /tmp/t8011-e7b8aa.o /tmp/s8003-c417eb.o /tmp/t8012-b08f91.o /tmp/s5l8960-f13494.o /tmp/t8010-09a3b9.o /tmp/s8001-ca072a.o /tmp/t7001-1ee25c.o /tmp/t8015-d0d506.o /tmp/synopsys_otg-7f7a31.o /tmp/mipi-e3a965.o /tmp/hal-22f392.o /tmp/aes_a7-db0cd6.o /tmp/aes_a9-8b6506.o /tmp/aes-c75318.o /tmp/sep-40be91.o /tmp/timer-4b33ed.o /tmp/recfg-5c8685.o /tmp/recfg_soc-c00261.o /tmp/gpio-928e5a.o /tmp/uart-7aa283.o /tmp/xnu-459cb1.o /tmp/sep_racer-339c1f.o /tmp/fdt-6890b1.o /tmp/fdt_rw-3fe16e.o /tmp/fdt_sw-eb377f.o /tmp/fdt_empty_tree-896ac0.o /tmp/fdt_overlay-1f3bb5.o /tmp/fdt_wip-a1d318.o /tmp/fdt_strerror-44726e.o /tmp/fdt_ro-0494c0.o /tmp/fdt_addresses-34a9e6.o /tmp/linux-a05db1.o /tmp/lzmadec-252084.o /tmp/DER_Decode-8c80fe.o /tmp/DER_Encode-ef373f.o /tmp/oids-763102.o /tmp/img4-7a2d7d.o -fatal_warnings -dead_strip -Z -lc -preload -no_uuid -e start -order_file src/sym_order.txt -image_base 0x100000000 -sectalign __DATA __common 0x8 -segalign 0x4000
Undefined symbols for architecture arm64:
  "___muloti4", referenced from:
      _reallocarray in libc.a(lib_a-reallocarray.o)
ld: symbol(s) not found for architecture arm64
clang: error: linker command failed with exit code 1 (use -v to see invocation)

aalyatim, 2022-04-24:
Any updates on this?

cakarlen, 2022-04-25:
Will PongoOS/jailbreak process work on iOS15 now with the more recent commits made on the iOS15 branch and compiling using those changes?

Siguza, 2022-04-27:
@aalyatim I experimented a bit more with clang-10. It's pretty broken, but you might just get this to compile if you manually pass the `libclang_rt.cc_kext_ios.a` lib to the linker (i.e. `PONGO_CFLAGS='-Wl,/usr/lib/llvm-10/lib/clang/10.0.0/lib/darwin/libclang_rt.cc_kext_ios.a'`).

@cakarlen No. There's still no public checkra1n that works with iOS 15 iBoot.