checkra1n/PongoOS #132: How to compile fdt module from `pongoOS/src/modules/linux/libfdt` without errors/correctly?

cuj, 2022-05-31:
Hi, I created an issue like almost a year ago ( #79 ) and I did not even get any answers, but I found a way, which returns errors. I tried on ubuntu linux:
```
$ git clone https://www.github.com/checkra1n/PongoOS.git
[... git output ...]
$ cd PongoOS/src/modules/linux/libfdt
$ cc fdt.c -I . -o fdt.bin
  /usr/bin/ld: /usr/lib/gcc/x86_64-linux-gnu/9/../../../x86_64-linux-gnu/Scrt1.o: in function `start':
  (.text+0x24): undefined reference to `main'
collect2: error: ld returned 1 exit status
$ # Same thing happens with gcc.
$ make
$ make: *** No targets specified and no makefile found.  Stop.
```

The reason why I wanted to compile fdt is so that I can load the `fdt` module to the pongoOS to boot linux/android sandcastle in my iPhone 7 Plus ([this build](https://assets.checkra.in/downloads/sandcastle/0175ae56bcba314268d786d1239535bca245a7b126d62a767e12de48fd20f470/linux-sandcastle.zip) in my case).

`fdt` used to come with pongoOS from checkra1n version 0.9.8.2 to checkra1n 0.10.2, but now it's not in pongoOS from checkra1n version 0.11.0 to version 0.12.4. It is needed to boot android/linux sandcastle on A10 devices.

So can anyone please help me compile the `fdt` module from `pongoOS/src/modules/linux/libfdt` without errors/correctly, so that I can load the module to my iPhone 7 Plus in pongoOS to boot it to android/linux sandcastle, even in the latest checkra1n/pongoOS versions? Thanks.

cuj, 2022-05-31:
@Siguza My apologies.
As I said in my post I opened an issue almost a year ago, which means I forgotten most of the pongoOS stuff, and fdt wasn't in some newer checkra1n versions than 0.10.2 and I assumd that fdt will never return even in version 0.12.4. closing this issue.

Siguza, 2022-05-31:
It's not a Pongo module, so you can't compile it that way. You'd have to add a whole bunch of module export functions. But what's the point? You can just recompile Pongo. What makes you even thing fdt is the issue? "Doesn't work" doesn't mean anything.

closed 2024-07-01 by cuj