checkra1n/PongoOS #137: A10 (iPad 7) sep pwn not working

nick-botticelli, 2022-08-04:
I have attempted to run blackbird exploit (`sep pwn`) on j171ap (A10 T8010) however I am met with a familiar "Registers are locked" and am unable to decrypt kbags. I receive the same message with `sep tz0_blackbird`.

It was my understanding that A10 was supported, yet I have tried running `sep pwn` immediately once booted into PongoOS and have received the same error message (both with PongoOS shipped with checkra1n 0.12.4 and built by myself on 4c9b7541).

Siguza, 2022-08-19:
The trustzone unlock patch is only applied on iOS 14.x. It sounds like you're below 14.0?

nick-botticelli, 2022-08-23:
> The trustzone unlock patch is only applied on iOS 14.x. It sounds like you're below 14.0?

Yes, I am. I assume then it isn't compatible (or is it just not enabled)?