checkra1n/PongoOS #139: *Feature Request* Downgrade Compatibility

bthedorff, 2022-08-26:
*This is a feature request however, it may not be 100% on the checkra1n team. At the very least, they could help and work with the other developer/s with the checkra1n portion.

Currently, there are a few ways to do a tethered downgrade checkm8 compatible phones to SEP compatible firmware versions without having any blobs. In my case, an iPhone 8. It is known that any tethered downgrade currently can't be jailbroken using checkra1n. The most well-known program is Sunst0rm (Github link [Here](https://github.com/mineek/sunst0rm)). I don't know how much work it would be for either side to get checkra1n to work with a downgraded OS but it would be a great thing for the Jailbreaking community to have the option for the most stable jailbreak. Since anyone with a tethered downgrade already has to use a PC to boot, we could (in theory) hopefully be able to boot the phone straight into a jailbroken state with checkra1n. It would be amazing if the checkra1n team could work with the developer/s to make it work!

Thank you for your time!

Arna13, 2022-08-27:
+1 for this!

Tenfortytwo, 2022-10-29:
unfortunately not man, Checkra1n is the only jailbreak that has worked successfulySent from my Huawei phone-------- Original message --------From: edwin ***@***.***>Date: Sat, Oct 29, 2022, 18:09To: checkra1n/PongoOS ***@***.***>Cc: Subscribed ***@***.***>Subject: Re: [checkra1n/PongoOS] *Feature Request* Downgrade Compatibility (Issue #139)
@bthedorff hey my bro, could you jailbreak your device with unc0ver? because i downgrade to ios 14.3 to jailbreak that version with taurine however taurine said that it is already jailbroken, i think it is because amfi patcher is done

edwin170, 2022-10-29:
mmm i tried to jailbreak with unc0ver however when i push the buttom jailbreak the application exit automatically