checkra1n/PongoOS #141: dualboot 14-15 using checkra1n can jailbreak my second os 14 ?

edwin170, 2022-10-29:
I dualbooted my device which is iPhone 6s in ios 15.7 with dualboot 14.3 and I want to know if pongo os could be modified in order to boot the second system disk0s1s8 as a jailbreak?

edwin170, 2022-10-29:
@Siguza, can checkra1n do that ?

Siguza, 2022-10-30:
Assuming you know how to operate pongoterm, try `xargs rootdev=disk0s1s8`.

edwin170, 2022-10-30:
xd :( thanks for your answer 

> Assuming you know how to operate pongoterm, try `xargs rootdev=disk0s1s8`.
> 
> Edit: I guess you'll need to switch in a new kernel, huh? Yeah, that's gonna take a lot more work...

:( xd, thanks for your answer my bro

edwin170, 2022-10-30:
@Siguza oh, I have 2 partitions to boot my second os which are disk0s1s8 (systemB) and disk0s1s9 (DataB) therefore I edit devicetree to mount the second data which is disk0s1s9 so xargs rootdev=disk0s1s8 in order to mount system and how can I mount datab with pongoterm?

Siguza, 2022-10-30:
You can't. That's not controlled at the kernel level, but from userspace. One of the boot-tasks spawned by launchd is responsible for mounting the data partition.

edwin170, 2022-10-30:
> You can't. That's not controlled at the kernel level, but from userspace. One of the boot-tasks spawned by launchd is responsible for mounting the data partition.

well, I had to patch devicetree (https://www.theiphonewiki.com/wiki/DeviceTree)with https://github.com/Ralph0045/dtree_patcher in order to mount datab, if I did not do that, the boot would mount disk0s1s2 instead disk0s1s9 so I did not need to edit or do something with launchd or the kernel, I think that would mount disk0s1s9 if we could load the image devicetree patched in pongoterm

edwin170, 2022-10-30:
> well, I had to patch devicetree (https://www.theiphonewiki.com/wiki/DeviceTree)with https://github.com/Ralph0045/dtree_patcher in order to mount datab, if I did not do that, the boot would mount [...]

@Siguza

edwin170, 2022-10-31:
> > > You can't. That's not controlled at the kernel level, but from userspace. One of the boot-tasks spawned by launchd is responsible for mounting the data partition.
> > 
> > 
> > well, I had to patch devicetree (https://www.theiphonewiki.com/wiki/DeviceTree)with https://github.com/Ralph0045/dtree_patcher in order to mount datab, if I did not do that, the boot would mount disk0s1s2 instead disk0s1s9 so I did not need to edit or do something with launchd or the kernel, I think that would mount disk0s1s9 if we could load the image devicetree patched in pongoterm
> 
> @Siguza so sir i think that devicetree say to lauchd what data partition will be mounted

edwin170, 2022-10-31:
when i tried to boot with checkra1n On ios 15.7 said (right before trigger) however I know the jailbreak does not work on ios 15.7 but i only want to boot a pongoterm, how can i boot pongoterm on ios 15.7?

closed 2022-11-03 by edwin170