checkra1n/PongoOS #142: Error on compile

Cozmo007, 2022-10-29:
[image]

Siguza, 2022-10-29:
`EMBEDDED_CFLAGS='-Wno-error'`

iamjeremybe, 2022-10-30:
I'm getting the same error, Ubuntu 22.04.1. I saw proposed solutions in other open issues about using various flags, including the ones suggested above, but 'make all' ends either with the errors above, or with this one, which has also been noted in other issues. Haven't found a combo of flags that works for me yet.

```
src/kernel/mm.c:378:14: warning: variable 'vm_index_start' set but not used [-Wunused-but-set-variable]
    uint32_t vm_index_start = 0;
             ^
src/kernel/mm.c:938:10: warning: variable 'is_tt1' set but not used [-Wunused-but-set-variable]
    bool is_tt1 = false;
         ^
2 warnings generated.
ld: could not process llvm bitcode object file, because /usr/bin/../lib/llvm/libLTO.so could not be loaded file '/tmp/stage3-25a83d.o' for architecture arm64
```

Ran 'make distclean' followed by 'EMBEDDED_CFLAGS='-Wno-error' make clean all'. Output is attached.
[compile_scroll.txt](https://github.com/checkra1n/PongoOS/files/9896243/compile_scroll.txt)

Siguza, 2022-10-30:
@iamjeremybe this means there's an issue with your toolchain. Check what version of LLVM your ld64 is built against, and if that is older than the version of clang you're using, rebuild ld64.

iamjeremybe, 2022-11-02:
I'm still learning, so thanks for your patience. How do I check these things?