checkra1n/PongoOS #144: checkra1n app missing

blackndoor, 2022-11-25:
The checkra1n app is not there after a success jailbreak with checkra1n macOS app.
I also tried to do as follow :

```
$ /Applications/checkra1n.app/Contents/MacOS/checkra1n -k ../../PongoOS/build/Pongo.bin
```

Then with pongoterm

```
sep auto
/send ../../PongoOS/build/checkra1n-kpf-pongo
modload
/send ../rdsk/rdsk.dmg // extracted from the checkra1n binary
ramdisk
xargs rootdev=md0
bootx
```

The iPhone boot correctly. The jailbreak seems working as cydia / newterm / ssh are working. But no checkra1n app.

Siguza, 2022-11-28:
This means the loader.dmg was not actually mounted. This is just a 0-sized file that would normally be hidden under the real mount (I don't quite remember why we create it). Is `/binpack` even mounted correctly?

blackndoor, 2022-11-29:
yes ls ```/binpack``` was mounted. I finaly get it work.

I used checkra1n to jailbreak, got the checkra1n.dmg from the device and got the loader.dmg from it.
I extracted the loader.dmg in ```/var/binpack/Application```and the checkra1n app was working.
I removed the jailbreak using the app and then jailbreak again using checkra1n.

I lost all my tweaks but at least now it works.
I guess I have to figure out how you push the checkra1n.dmg to the device and used it.