checkra1n/PongoOS #145: Cannot build on macOS 13

alfiecg24, 2022-11-27:
I cannot seem to build this repository on macOS Ventura.

Steps to reproduce:

- Clone repository
- Run `make all` (gives unused variable errors)
- Run `make all EMBEDDED_CC_FLAGS="-Wno-error"`
- Still doesn't build

Running `make all EMBEDDED_CC_FLAGS="-Wno-error"` without running `make all` first also fails when compiling newlib with the message `"../stdio/lib.a": No such file or directory`.

Running the commands above gives many errors to do with implicit declarations of `iprintf` and `siprintf` being invalid in C99. I am using the regular clang, gcc etc. from /usr/bin if that helps. Hope I can be of assistance if you need it.

alfiecg24, 2022-11-27:
Update: adding `-Wno-implicit-function-declaration` to the flags fails to build with these errors

``` src/kernel/support/sbrk.c:33:1: error: unknown type name 'caddr_t'
caddr_t _sbrk(int size) {
^
src/kernel/support/sbrk.c:42:13: error: use of undeclared identifier 'caddr_t'
    return (caddr_t)cursor_copy;
            ^
2 errors generated.

Siguza, 2022-11-27:
It should be `EMBEDDED_CFLAGS`, not `EMBEDDED_CC_FLAGS`. If you redefine the latter, you wipe out all the default flags.

alfiecg24, 2022-11-27:
Ah okay, I must have misread an answer on another issue. Thanks for the help - issue is solved.

closed 2022-11-27 by alfiecg24