checkra1n/PongoOS #146: Running fetch_stdout.py triggers kernel panic

alfiecg24, 2022-11-27:
I ran the CLI version of checkra1n with the command `checkra1n -cpvV`
I then did `python3 fetch_stdout.py` and got the error message `usb.core.USBTimeoutError: [Errno 60] Operation timed out` and no output
Running it again panics the kernel and pongoOS displays the message
```
panic: USB BUG: 0x73656e642034
crashed task: usb

crashed process: kernel

alfiecg24, 2022-11-27:
I have the call stack if needed:

```
Call stack: 0xfffffffe000b76c0: fp 0xfffffffe000b76e0. lr 0x000000010001fd6c
Call stack: 0xfffffffe000b76e0: fp 0xfffffffe000b77f0. lr 0x0000000100020834
Call stack: 0xfffffffe000b77f0: fp 0x0000000000000000. lr 0x0000000100011870

Ijustplaymc, 2023-01-07:
Encountered same issue, same call stack

fetch_stdout works fine on linux, but if I issue a command (like help) then try to dump it it kernel panics

os : linux mint
iPhone SE, 14.1
Checkra1n 0.12.4
x86_64 device
libusb/pyusb and other dependencies work fine
standard usb a cable, no adapters, also tried with 3rd party