checkra1n/PongoOS #161: ld: symbol(s) not found for architecture arm64

amats211, 2023-04-18:
hi, trying to build in macboot pro mid2012 catalina and i get errors after i run make all.

> Undefined symbols for architecture arm64:
>   "___muloti4", referenced from:
>       _reallocarray in cc-8da2a4.o
> ld: symbol(s) not found for architecture arm64
> clang: error: linker command failed with exit code 1 (use -v to see invocation)
> make: *** [build/Pongo] Error 1 

what can i do to fix?

Siguza, 2023-04-18:
It means you're missing libclang_rt for iOS... what toolchain are you using?

amats211, 2023-04-19:
i am not sure if i have installed this .. 
https://developer.arm.com/downloads/-/arm-gnu-toolchain-downloads

any way to check?

Siguza, 2023-04-19:
What does `clang -v` print?

amats211, 2023-04-20:
mac@Macs-MacBook-Pro bin % clang -v
Apple clang version 12.0.0 (clang-1200.0.32.29)
Target: x86_64-apple-darwin19.6.0
Thread model: posix
InstalledDir: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin

amats211, 2023-07-25:
Hi everyone, just updating this as i still experience the issue.. Error only occurs on iOS15 branch, on main branch my mac can make with no errors. 

So this should be the correct output from clang -v and not the one i posted above hehe

"/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/ld" -demangle -object_path_lto /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/cc-c2672b.o -lto_library /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/lib/libLTO.dylib -static -arch arm64 -platform_version ios 12.0.0 14.4 -syslibroot /Applications/Xcode.app/Contents/Developer/Platforms/iPhoneOS.platform/Developer/SDKs/iPhoneOS14.4.sdk -mllvm -enable-machine-outliner -mllvm -enable-linkonceodr-outlining -o build/Pongo -Lnewlib/aarch64-none-darwin/fixup /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/demote_patch-cb6db6.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/entry-fbfb75.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/jump_to_image-bbbdc9.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/patches-c06988.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/int-bdb7a9.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/sep_racer-2172fa.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/xnu-0a7c43.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/main-ca953c.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/stage3-16a2ed.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/modload-134638.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/modload_macho-512075.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/entry-401070.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/legacy-01d0c8.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/locks-295909.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/lowlevel-ef7a22.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/main_task-6bf8b5.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/mm-55944f.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/panic-0f4cbe.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/syscall-88a47e.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/task-0ff003.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/vfs-ceba25.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/command-e4a522.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/hex-1bad67.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/linux-a1d0ab.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/main-ab81e5.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/usbloader-639d9e.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/aes-b475ff.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/aes_a7-19402b.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/aes_a9-6987af.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/dt-778fd6.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/dt_get-66ac15.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/fb-d9c197.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/fuse-91925d.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/gpio-f73b9b.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/hal-3c6dac.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/mipi-6b1370.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/s5l8960-62879c.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/s8000-9605e7.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/s8001-08bd60.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/s8003-4552a3.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/t7000-15ff3b.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/t7001-c42e86.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/t8010-392cc1.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/t8011-81910a.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/t8012-47de67.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/t8015-5074ed.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/recfg-beb23a.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/recfg_soc-d5f667.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/sep-46023c.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/timer-2458df.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/tz-ab4694.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/uart-3a83ba.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/synopsys_otg-819349.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/xnu-140cb9.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/exit-6b2130.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/io-f389e4.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/malloc-fcf2f9.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/process-e38f4e.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/sbrk-119a26.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/img4-c4e6cd.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/DER_Decode-775790.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/DER_Encode-a25c07.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/oids-40433d.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/lzmadec-3da5bd.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/linux-e22c07.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/fdt-769a67.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/fdt_addresses-9ae1e4.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/fdt_empty_tree-46c876.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/fdt_overlay-f163db.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/fdt_ro-fc0915.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/fdt_rw-16cbeb.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/fdt_strerror-9d8ac2.o /var/f [... cut ...] T/fdt_sw-a7ff0e.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/fdt_wip-3b163d.o -dead_strip -Z -lc -preload -no_uuid -e start -order_file src/sym_order.txt -image_base 0x100000000 -sectalign __DATA __common 0x8 -segalign 0x4000
Undefined symbols for architecture arm64:
  "___muloti4", referenced from:
      _reallocarray in cc-c2672b.o
ld: symbol(s) not found for architecture arm64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make: *** [build/Pongo] Error 1

amats211, 2023-07-25:
i tried to change toolchain, used swift but still same.. is it because of the 14.4 sdk? 

clang -cc1 version 13.0.0 based upon LLVM 13.0.0 default target x86_64-apple-darwin19.6.0
ignoring nonexistent directory "include/modules/linux/"
#include "..." search starts here:
#include <...> search starts here:
 newlib/aarch64-none-darwin/include
 src/lib
 include
 apple-include
 src/kernel
 src/drivers
 src/modules/linux/libfdt
 /Library/Developer/Toolchains/swift-5.8.1-RELEASE.xctoolchain/usr/lib/clang/13.0.0/include
End of search list.
 "/usr/bin/ld" -demangle -object_path_lto /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/cc-0d86a1.o -lto_library /Library/Developer/Toolchains/swift-5.8.1-RELEASE.xctoolchain/usr/lib/libLTO.dylib -static -arch arm64 -platform_version ios 12.0.0 14.4 -syslibroot /Applications/Xcode.app/Contents/Developer/Platforms/iPhoneOS.platform/Developer/SDKs/iPhoneOS14.4.sdk -mllvm -enable-machine-outliner -mllvm -enable-linkonceodr-outlining -o build/Pongo -Lnewlib/aarch64-none-darwin/fixup /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/demote_patch-d6f772.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/entry-809ea5.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/jump_to_image-946dc9.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/patches-5dbae7.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/int-cc9237.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/sep_racer-5a563a.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/xnu-07114d.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/main-be1464.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/stage3-a404f7.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/modload-96f75b.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/modload_macho-10cc00.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/entry-e54c66.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/legacy-2d04be.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/locks-0be1ae.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/lowlevel-66540c.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/main_task-a0ba19.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/mm-b86fca.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/panic-76fc5e.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/syscall-24bd0e.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/task-a3994b.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/vfs-a450dc.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/command-166593.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/hex-2e695e.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/linux-793574.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/main-06c77c.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/usbloader-6eb502.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/aes-6cb05b.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/aes_a7-b6a4a8.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/aes_a9-3f68d7.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/dt-13b923.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/dt_get-a87411.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/fb-29562e.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/fuse-79cdd1.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/gpio-7f633e.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/hal-fac705.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/mipi-0703fe.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/s5l8960-c2ae4c.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/s8000-e53caa.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/s8001-83f172.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/s8003-c88581.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/t7000-d3a9e3.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/t7001-3c0a7f.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/t8010-3eddd0.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/t8011-55c200.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/t8012-51c2d0.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/t8015-1983fb.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/recfg-fb0fa2.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/recfg_soc-cbdbe7.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/sep-4fe184.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/timer-b572a5.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/tz-3bdb97.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/uart-a65fc2.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/synopsys_otg-bd4a45.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/xnu-34dd49.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/exit-826c97.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/io-ab8386.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/malloc-541060.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/process-eebd56.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/sbrk-4495ef.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/img4-17f9af.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/DER_Decode-b503a5.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/DER_Encode-0ee9ef.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/oids-c48a29.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/lzmadec-2bef4e.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/linux-3c8797.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/fdt-89bd0e.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/fdt_addresses-39ed52.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/fdt_empty_tree-dc7e05.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/fdt_overlay-84848a.o / [... cut ...] dt_sw-18f0ad.o /var/folders/q7/3yfmbmfn0ql0lcz29y726dq80000gn/T/fdt_wip-6dd6e8.o -dead_strip -Z -lc -preload -no_uuid -e start -order_file src/sym_order.txt -image_base 0x100000000 -sectalign __DATA __common 0x8 -segalign 0x4000
Undefined symbols for architecture arm64:
  "___muloti4", referenced from:
      _reallocarray in cc-0d86a1.o
ld: symbol(s) not found for architecture arm64
clang-13: error: linker command failed with exit code 1 (use -v to see invocation)
make: *** [build/Pongo] Error 1

markwjkelly, 2024-08-03:
Anyone have a resolution for this issue.  I have something very similar.  Everything worked fine until xcode update ... now getting ...
Undefined symbols for architecture arm64:
  "_ReloadPage", referenced from:
      _QuitApp_ReloadPage_m8D82739E5C8C05F16EBD6ECA2C5FF8A40A2B74E2 in libGameAssembly.a[3](v7wf8c39xmtf.o)
      _QuitApp_Reload_m302C2E7668DDEBFB2A15DF030C668E556CBE3CB7 in libGameAssembly.a[3](v7wf8c39xmtf.o)
ld: symbol(s) not found for architecture arm64
clang: error: linker command failed with exit code 1 (use -v to see invocation) 

THX in advance!