checkra1n/PongoOS #166: Ubuntu Build Error : ld: could not process llvm bitcode object file

Slayy2357, 2023-06-17:
I tried to build PongoOS on ubuntu 22.04 and this error happened : `ld: could not process llvm bitcode object file, because /usr/bin/../lib/llvm/libLTO.so could not be loaded file '/tmp/stage3-c42f2a.o' for architecture arm64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make: *** [Makefile:99 : build/Pongo] Erreur 1`

Here there is the complete return after make command (txt file because long): 
https://anonfiles.com/u9sdl9x5z0/Pongo_Error_txt
If someone can help me it can be great, thanks in advance !

Slayy2357, 2023-06-22:
Nobody had a solution ?

Siguza, 2023-06-22:
You need to build ld64 against the version of clang that you're using. You can use new ld64 with old clang, but not the other way round.

I've been meaning to update the build on the checkra1n repo, but... time.

Slayy2357, 2023-06-23:
Thanks for the reply, i will test this x)

Slayy2357, 2023-06-25:
But can you send me the ld64 repo that i need to build ? I think i don't have found the good repo to build...

Siguza, 2023-06-26:
I've been using this: https://github.com/Siguza/ld64

The real work has been done in tpoechtrager/cctools-port though. Unfortunately that hasn't been updated to the latest ld64 source yet, but I'm hoping it'll work on newest LLVM anyway.  
I'm planning to clean up the toolchain situation, among many other things, and have prebuilt binaries for arm64 and x86_64 Debian. But it all takes time...

Slayy2357, 2023-06-27:
Thanks, but it was the repo that i have tested so the make command is not working, i got errors... Could you just tell me with which version of llvm you managed to compile it?

51mpl1, 2023-07-09:
Just for the information and maybe for version-checking. Got exactly the same issue with
 debian 12,
 clang version 14.0.6,
 ld64 530-2,
 cctools-strip 949.0.1-2.

Slayy2357, 2023-08-05:
Did you found an solution ?

polypolyman, 2023-09-17:
I had to add `LD_LIBRARY_PATH="/usr/lib/llvm-11/lib/"` to the commandline for make (before make). Your command may vary depending on the location of your llvm libraries (i.e. a different version will have a different number there, etc.)

jacquesTech, 2023-09-17:
Try update your shell command lines , make sure you run root modr