checkra1n/PongoOS #169: Android on iOS15 branch currently broken

zoe-vb, 2023-07-12:
Hi after building the latest version of the iOS 15 branch the linux kernel provided by corellium boots correctly on the iPhone 7 however the Android kernel does not and results in a crash before executing the kernel.  Just wondering if you guys were looking into this or if you had decided that it wasn't worth fixing?

alfiecg24, 2023-07-16:
If the Linux kernel boots okay, this is probably an issue with the Android kernel, not PongoOS. Hope this helps.

zoe-vb, 2023-07-16:
@alfiecg24 I have attempted to compile the sandcastle linux kernel again from source by following the exact instructions and it fails with the same issue that the sandcastle android one did which suggested that it was possibly a pongo issue. I also tried the linux-apple fork of linux and received the same result. Is there any way which you would recommend that I compile the kernel because the method suggested isn't working for me.

alfiecg24, 2023-07-16:
I’m not too well-versed with Linux/Android through PongoOS, so unfortunately can’t offer much help on the compiling side of things. What’s the error that you receive?

zoe-vb, 2023-07-16:
@alfiecg24 

[image]

alfiecg24, 2023-07-16:
Hmm, this doesn’t make it obvious as to what went wrong. Are you sure the kernel image is decompressed? I don’t have an iPhone 7, but I do have an A10X iPad, so it may be possible for me to look into this at some point.

However, if I remember correctly, Sandcastle doesn’t work on iOS 15 or 16, and may be for other reasons which are beyond my understanding. Some people have managed to boot Linux on iOS 15/16 though, so that could be a starting point.

If it is indeed a problem with Sandcastle and not PongoOS, in order to boot Sandcastle, it will probably require a large amount of work of which I definitely would be out of my depth with. It ultimately depends on whether you are set on booting Android only or whether Linux is an alternate option.

zoe-vb, 2023-07-16:
@alfiecg24 I'm more interested in linux anyway, I haven't been able to find anything about linux on iOS 15, could you link me to some of those people / the posts?

alfiecg24, 2023-07-16:
Here’s a good example: https://github.com/asdfugil/linux-apple. It may or may not work out of the box, but it did at one point involve chain loading a PongoOS versions in order to boot it. I’ll see if I can look into it in the next few days, but I did get linux-apple working on iOS 14 last year, so it may be relatively similar. I’ll update you if I have any progress.

zoe-vb, 2023-07-16:
@alfiecg24 so I had a go at using the kernel for 16k and the dtbpack from the github actions and it's the same error, I think its a pongoOS issue. @konradybcio had a fork of PongoOS also for the original linux-apple that included patches to make linux actually work but those were on the old master branch and not on the current one.

alfiecg24, 2023-07-17:
[This](https://github.com/SoMainline/linux-apple-resources/blob/master/HOWTO.md) is the guide I used last time, which indicates you need a Linux machine to build the kernel for Linux on iOS - which I unfortunately do not own. You can try following this guide and see if it still works with the updated forks. The same person from the linux-apple fork above has a fork of konradybcio's PongoOS [here](https://github.com/asdfugil/PongoOS), could work too.

zoe-vb, 2023-07-17:
@alfiecg24 Their fork of Pongo is a direct fork of the patched on and has the same issue of being patched only on the master branch, I'm going to attempt to patch the iOS15 branch with the patches from the master branch and see if that works. I have previously used the So-Mainline guide and it works perfectly on my iPhone 6 on iOS 12 and my SE on iOS 14. I'll update and let you know how it goes

alfiecg24, 2023-07-17:
I wish you luck in doing so - it may be important to bear in mind that the iOS15 branch of that fork has specific improvements for Linux, so the reason that it doesn’t have that patch could’ve be that it’s not needed for iOS 15.

UnCor3, 2024-08-23:
@zoe-vb Have you been able to figure out ? I went with steps [here](https://github.com/SoMainline/linux-apple-resources/blob/master/HOWTO.md) but it didn't work I even tried very old versions of pongoOS nothing seems to work