checkra1n/PongoOS #185: Can't compile pongoOS

slayy2357, 2024-07-10:
ld: warning: ignoring file /tmp/stage3-150467.o, building for free standing-arm64 but attempting to link with file built for unknown-unsupported file format ( 0xDE 0xC0 0x17 0x0B 0x00 0x00 0x00 0x00 0x14 0x00 0x00 0x00 0x74 0x24 0x00 0x00 )
ld: warning: ignoring file /tmp/main-c2dd97.o, building for free standing-arm64 but attempting to link with file built for unknown-unsupported file format ( 0xDE 0xC0 0x17 0x0B 0x00 0x00 0x00 0x00 0x14 0x00 0x00 0x00 0x74 0x0B 0x00 0x00 )
ld: warning: ignoring file /tmp/dtree_getprop-6f1dfc.o, building for free standing-arm64 but attempting to link with file built for unknown-unsupported file format ( 0xDE 0xC0 0x17 0x0B 0x00 0x00 0x00 0x00 0x14 0x00 0x00 0x00 0xC8 0x11 0x00 0x00 )
ld: warning: ignoring file /tmp/dtree-b1538d.o, building for free standing-arm64 but attempting to link with file built for unknown-unsupported file format ( 0xDE 0xC0 0x17 0x0B 0x00 0x00 0x00 0x00 0x14 0x00 0x00 0x00 0x10 0x17 0x00 0x00 )
ld: warning: ignoring file /tmp/entry-479710.o, building for free standing-arm64 but attempting to link with file built for unknown-unsupported file format ( 0xDE 0xC0 0x17 0x0B 0x00 0x00 0x00 0x00 0x14 0x00 0x00 0x00 0xD0 0x36 0x00 0x00 )
ld: warning: ignoring file /tmp/legacy-40ff52.o, building for free standing-arm64 but attempting to link with file built for unknown-unsupported file format ( 0xDE 0xC0 0x17 0x0B 0x00 0x00 0x00 0x00 0x14 0x00 0x00 0x00 0x58 0x0F 0x00 0x00 )
ld: warning: ignoring file /tmp/lowlevel-a994e1.o, building for free standing-arm64 but attempting to link with file built for unknown-unsupported file format ( 0xDE 0xC0 0x17 0x0B 0x00 0x00 0x00 0x00 0x14 0x00 0x00 0x00 0xAC 0x6F 0x00 0x00 )
ld: warning: ignoring file /tmp/main_task-535ff7.o, building for free standing-arm64 but attempting to link with file built for unknown-unsupported file format ( 0xDE 0xC0 0x17 0x0B 0x00 0x00 0x00 0x00 0x14 0x00 0x00 0x00 0xFC 0x14 0x00 0x00 )
ld: warning: ignoring file /tmp/locks-67af31.o, building for free standing-arm64 but attempting to link with file built for unknown-unsupported file format ( 0xDE 0xC0 0x17 0x0B 0x00 0x00 0x00 0x00 0x14 0x00 0x00 0x00 0x88 0x11 0x00 0x00 )
ld: warning: ignoring file /tmp/mm-1ea08d.o, building for free standing-arm64 but attempting to link with file built for unknown-unsupported file format ( 0xDE 0xC0 0x17 0x0B 0x00 0x00 0x00 0x00 0x14 0x00 0x00 0x00 0xAC 0x88 0x00 0x00 )
ld: warning: ignoring file /tmp/task-ad3acf.o, building for free standing-arm64 but attempting to link with file built for unknown-unsupported file format ( 0xDE 0xC0 0x17 0x0B 0x00 0x00 0x00 0x00 0x14 0x00 0x00 0x00 0xF4 0x87 0x00 0x00 )
ld: warning: ignoring file /tmp/panic-8f5c2b.o, building for free standing-arm64 but attempting to link with file built for unknown-unsupported file format ( 0xDE 0xC0 0x17 0x0B 0x00 0x00 0x00 0x00 0x14 0x00 0x00 0x00 0x64 0x16 0x00 0x00 )
ld: warning: ignoring file /tmp/syscall-64c6c7.o, building for free standing-arm64 but attempting to link with file built for unknown-unsupported file format ( 0xDE 0xC0 0x17 0x0B 0x00 0x00 0x00 0x00 0x14 0x00 0x00 0x00 0xAC 0x15 0x00 0x00 )
ld: warning: ignoring file /tmp/exit-dc241e.o, building for free standing-arm64 but attempting to link with file built for unknown-unsupported file format ( 0xDE 0xC0 0x17 0x0B 0x00 0x00 0x00 0x00 0x14 0x00 0x00 0x00 0x94 0x09 0x00 0x00 )
ld: warning: ignoring file /tmp/io-7bc058.o, building for free standing-arm64 but attempting to link with file built for unknown-unsupported file format ( 0xDE 0xC0 0x17 0x0B 0x00 0x00 0x00 0x00 0x14 0x00 0x00 0x00 0x18 0x21 0x00 0x00 )
ld: warning: ignoring file /tmp/vfs-c1db2a.o, building for free standing-arm64 but attempting to link with file built for unknown-unsupported file format ( 0xDE 0xC0 0x17 0x0B 0x00 0x00 0x00 0x00 0x14 0x00 0x00 0x00 0xDC 0x16 0x00 0x00 )
ld: warning: ignoring file /tmp/malloc-250b11.o, building for free standing-arm64 but attempting to link with file built for unknown-unsupported file format ( 0xDE 0xC0 0x17 0x0B 0x00 0x00 0x00 0x00 0x14 0x00 0x00 0x00 0x54 0x0A 0x00 0x00 )
ld: warning: ignoring file /tmp/sbrk-159d1f.o, building for free standing-arm64 but attempting to link with file built for unknown-unsupported file format ( 0xDE 0xC0 0x17 0x0B 0x00 0x00 0x00 0x00 0x14 0x00 0x00 0x00 0xE0 0x0C 0x00 0x00 )
ld: warning: ignoring file /tmp/process-706be8.o, building for free standing-arm64 but attempting to link with file built for unknown-unsupported file format ( 0xDE 0xC0 0x17 0x0B 0x00 0x00 0x00 0x00 0x14 0x00 0x00 0x00 0x40 0x0D 0x00 0x00 )
ld: warning: ignoring file /tmp/autoboot-f8bf67.o, building for free standing-arm64 but attempting to link with file built for unknown-unsupported file format ( 0xDE 0xC0 0x17 0x0B 0x00 0x00 0x00 0x00 0x14 0x00 0x00 0x00 0xC4 0x0D 0x00 0x00 )
ld: warning: ld: warning: ignoring file /tmp/command-cf3c34.o, building for free standing-arm64 but attempting to link with file built for unknown-unsupported file format ( 0xDE 0xC0 0x17 0x0B 0x00 0x00 0x00 0x00 0x14 0x00 0x00 0x00 0x8C 0x27 0x00 0x00 )
ld: warning: ignoring file /tmp/modload_macho-1fe64d.o, building for free standing-arm64 but attempting to link with file built for unknown-unsupported file format ( 0xDE 0xC0 0x17 0x0B 0x00 0x00 0x00 0x00 0x14 0x00 0x00 0x00 0xB4 0x21 0x00 0x00 )ld: warning: ignoring file /tmp/main-914ede.o, building for free standing-arm64 but attempting to link with file built for unknown-unsupported file format ( 0xDE 0xC0 0x17 0x0B 0x00 0x00 0x00 0x00 0x14 0x00 0x00 0x00 0x50 0x3A 0x00 0x00 )
ld: warning: ignoring file /tmp/usbloader-59ca95.o, building for free standing-arm64 but attempting to link with file built for unknown-unsupported file format ( 0xDE 0xC0 0x17 0x0B 0x00 0x00 0x00 0x00 0x14 0x00 0x00 0x00 0x34 0x1F 0x00 0x00 )ld: warning: ignoring file /tmp/aes_a7-81cc73.o, building for free standing-arm64 but attempting to link with file built for unknown-unsupported file format ( 0xDE 0xC0 0x17 0x0B 0x00 0x00 0x00 0x00 0x14 0x00 0x00 0x00 0xEC 0x12 0x00 0x00 )ignoring file /tmp/modload-7e0756.o, building for free standing-arm64 but attempting to link with file built for unknown-unsupported file format ( 0xDE 0xC0 0x17 0x0B 0x00 0x00 0x00 0x00 0x14 0x00 0x00 0x00 0x4C 0xFA 0x05 0x00 )ld: warning: ignoring file /tmp/linux-1fe5b5.o, building for free standing-arm64 but attempting to link with file built for unknown-unsupported file format ( 0xDE 0xC0 0x17 0x0B 0x00 0x00 0x00 0x00 0x14 0x00 0x00 0x00 0x00 0x17 0x00 0x00 )
ld: warning: ld: warning: ignoring file /tmp/hex-2eb907.o, building for free standing-arm64 but attempting to link with file built for unknown-unsupported file format ( 0xDE 0xC0 0x17 0x0B 0x00 0x00 0x00 0x00 0x14 0x00 0x00 0x00 0xC8 0x12 0x00 0x00 )

ld: warning: ignoring file /tmp/aes-d1293a.o, building for free standing-arm64 but attempting to link with file built for unknown-unsupported file format ( 0xDE 0xC0 0x17 0x0B 0x00 0x00 0x00 0x00 0x14 0x00 0x00 0x00 0xAC 0x1F 0x00 0x00 )
ld: warning: 
ignoring file /tmp/fb-d95f6a.o, building for free standing-arm64 but attempting to link with file built for unknown-unsupported file format ( 0xDE 0xC0 0x17 0x0B 0x00 0x00 0x00 0x00 0x14 0x00 0x00 0x00 0xB8 0x33 0x00 0x00 )
ignoring file /tmp/aes_a9-3cd849.o, building for free standing-arm64 but attempting to link with file built for unknown-unsupported file format ( 0xDE 0xC0 0x17 0x0B 0x00 0x00 0x00 0x00 0x14 0x00 0x00 0x00 0x48 0x14 0x00 0x00 )
ld: warning: ignoring file /tmp/gpio-76f5f7.o, building for free standing-arm64 but attempting to link with file built for unknown-unsupported file format ( 0xDE 0xC0 0x17 0x0B 0x00 0x00 0x00 0x00 0x14 0x00 0x00 0x00 0xA4 0x11 0x00 0x00 )
ld: warning: ignoring file /tmp/mipi-0e79c6.o, building for free standing-arm64 but attempting to link with file built for unknown-unsupported file format ( 0xDE 0xC0 0x17 0x0B 0x00 0x00 0x00 0x00 0x14 0x00 0x00 0x00 0x00 0x18 0x00 0x00 )
ld: warning: ignoring file /tmp/s8000-204684.o, building for free standing-arm64 but attempting to link with file built for unknown-unsupported file format ( 0xDE 0xC0 0x17 0x0B 0x00 0x00 0x00 0x00 0x14 0x00 0x00 0x00 0x5C 0x10 0x00 0x00 )
ld: warning: ld: warning: ld: warning: ignoring file /tmp/s5l8960-2e326b.o, building for free standing-arm64 but attempting to link with file built for unknown-unsupported file format ( 0xDE 0xC0 0x17 0x0B 0x00 0x00 0x00 0x00 0x14 0x00 0x00 0x00 0x70 0x10 0x00 0x00 )
[... cut ...]
      start$l0 in entry-53a026.o
  "_enable_interrupts", referenced from:
      __task_switch_asserted in int-0ac665.o
      _task_load_asserted in int-0ac665.o
  "_sep_check_mailbox", referenced from:
      _sep_racer in sep_racer-38da58.o
  "_sep_fast_check_mailbox", referenced from:
      _sep_racer in sep_racer-38da58.o
ld: symbol(s) not found for architecture arm64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make: *** [Makefile:99 : build/Pongo] Erreur 1

Siguza, 2024-07-10:
Are you manually passing a compiler (`CLANG`/`EMBEDDED_CC`) or llvm-config (`LLVM_CONFIG`/`EMBEDDED_LLVM_CONFIG`)?

slayy2357, 2024-07-10:
> Are you manually passing a compiler (`CLANG`/`EMBEDDED_CC`) or llvm-config (`LLVM_CONFIG`/`EMBEDDED_LLVM_CONFIG`)?

I don't know, I have compiled ld64/cctool and copy to path, installed llvm, installed clang and then try to compile pongoOS and this error appened. So I think llvm-config ?

Siguza, 2024-07-10:
Alright, could you please post the output of:

```
llvm-config --bindir
```

and

```
find "$(llvm-config --bindir)/../lib" -name 'libLTO.*'
```

slayy2357, 2024-07-10:
> Alright, could you please post the output of:
> 
> ```
> llvm-config --bindir
> ```
> 
> and
> 
> ```
> find "$(llvm-config --bindir)/../lib" -name 'libLTO.*'
> ```

Of course :
llvm-config --bindir : 
`/usr/lib/llvm-18/bin`

find "$(llvm-config --bindir)/../lib" -name 'libLTO.*' : 
`/usr/lib/llvm-18/bin/../lib/libLTO.so.18.1
/usr/lib/llvm-18/bin/../lib/libLTO.so`

Siguza, 2024-07-11:
Ok, try adding this before your `make` invocation:

```
EMBEDDED_LDFLAGS="-fuse-ld=$(which ld64) -Wl,-lto_library,$(llvm-config --bindir)/../lib/libLTO.so"
```

slayy2357, 2024-07-11:
> Ok, try adding this before your `make` invocation:
> 
> ```
> EMBEDDED_LDFLAGS="-fuse-ld=$(which ld64) -Wl,-lto_library,$(llvm-config --bindir)/../lib/libLTO.so"
> ```

Sorry, it didn't work, probably the same error : 
EMBEDDED_LDFLAGS="-fuse-ld=$(which ld64) -Wl,-lto_library,$(llvm-config --bindir)/../lib/libLTO.so" make all
make  -C newlib all
make[1] : on entre dans le répertoire « /home/unknow/PongoOS/newlib »
make  -C /home/unknow/PongoOS/newlib/build all
make[2] : on entre dans le répertoire « /home/unknow/PongoOS/newlib/build »
make "AR_FLAGS=rc" "CC_FOR_BUILD=gcc" "CFLAGS=--target=arm64-apple-ios12.0 -Wall -O3 -ffreestanding -nostdlib -nostdlibinc -fno-builtin -fno-blocks -U__nonnull -D_LDBL_EQ_DBL " "CCASFLAGS=--target=arm64-apple-ios12.0 -Wall -O3 -ffreestanding -nostdlib -nostdlibinc -fno-builtin -fno-blocks -U__nonnull -D_LDBL_EQ_DBL " "CFLAGS_FOR_BUILD=" "CFLAGS_FOR_TARGET=" "INSTALL=/usr/bin/install -c" "LDFLAGS=" "LIBCFLAGS=" "LIBCFLAGS_FOR_TARGET=" "MAKE=make" "MAKEINFO=/bin/bash /home/unknow/PongoOS/newlib/src/missing --run makeinfo " "PICFLAG=" "PICFLAG_FOR_TARGET=" "SHELL=/bin/bash" "EXPECT=`if test -f ./../expect/expect; then echo ./../expect/expect; else echo expect; fi`" "RUNTEST=`if test -f /home/unknow/PongoOS/newlib/src/newlib/../dejagnu/runtest; then echo /home/unknow/PongoOS/newlib/src/newlib/../dejagnu/runtest; else echo runtest; fi`" "RUNTESTFLAGS=" "exec_prefix=/home/unknow/PongoOS/newlib" "infodir=/home/unknow/PongoOS/newlib/share/info" "libdir=/home/unknow/PongoOS/newlib/lib" "prefix=/home/unknow/PongoOS/newlib" "tooldir=/home/unknow/PongoOS/newlib/aarch64-none-darwin" "top_toollibdir=/home/unknow/PongoOS/newlib/aarch64-none-darwin/lib" "AR=llvm-ar" "AS=as" "CC=clang -I/home/unknow/PongoOS/newlib/build/targ-include -I/home/unknow/PongoOS/newlib/src/newlib/libc/include" "LD=" "LIBCFLAGS=" "NM=" "PICFLAG=" "RANLIB=llvm-ranlib" "DESTDIR=" all-recursive
make[3] : on entre dans le répertoire « /home/unknow/PongoOS/newlib/build »
true "AR_FLAGS=rc" "CC_FOR_BUILD=gcc" "CFLAGS=--target=arm64-apple-ios12.0 -Wall -O3 -ffreestanding -nostdlib -nostdlibinc -fno-builtin -fno-blocks -U__nonnull -D_LDBL_EQ_DBL " "CCASFLAGS=--target=arm64-apple-ios12.0 -Wall -O3 -ffreestanding -nostdlib -nostdlibinc -fno-builtin -fno-blocks -U__nonnull -D_LDBL_EQ_DBL " "CFLAGS_FOR_BUILD=" "CFLAGS_FOR_TARGET=" "INSTALL=/usr/bin/install -c" "LDFLAGS=" "LIBCFLAGS=" "LIBCFLAGS_FOR_TARGET=" "MAKE=make" "MAKEINFO=/bin/bash /home/unknow/PongoOS/newlib/src/missing --run makeinfo  " "PICFLAG=" "PICFLAG_FOR_TARGET=" "SHELL=/bin/bash" "EXPECT=expect" "RUNTEST=runtest" "RUNTESTFLAGS=" "exec_prefix=/home/unknow/PongoOS/newlib" "infodir=/home/unknow/PongoOS/newlib/share/info" "libdir=/home/unknow/PongoOS/newlib/lib" "prefix=/home/unknow/PongoOS/newlib" "tooldir=/home/unknow/PongoOS/newlib/aarch64-none-darwin" "top_toollibdir=/home/unknow/PongoOS/newlib/aarch64-none-darwin/lib" "AR=llvm-ar" "AS=as" "CC=clang -I/home/unknow/PongoOS/newlib/build/targ-include -I/home/unknow/PongoOS/newlib/src/newlib/libc/include" "LD=" "LIBCFLAGS=" "NM=" "PICFLAG=" "RANLIB=llvm-ranlib" "DESTDIR=" DO=all multi-do # make
Making all in libc
make[4] : on entre dans le répertoire « /home/unknow/PongoOS/newlib/build/libc »
Making all in argz
make[5] : on entre dans le répertoire « /home/unknow/PongoOS/newlib/build/libc/argz »
make[5]: rien à faire pour « all ».
make[5] : on quitte le répertoire « /home/unknow/PongoOS/newlib/build/libc/argz »
Making all in stdlib
make[5] : on entre dans le répertoire « /home/unknow/PongoOS/newlib/build/libc/stdlib »
make[5]: rien à faire pour « all ».
make[5] : on quitte le répertoire « /home/unknow/PongoOS/newlib/build/libc/stdlib »
Making all in ctype
make[5] : on entre dans le répertoire « /home/unknow/PongoOS/newlib/build/libc/ctype »
make[5]: rien à faire pour « all ».
make[5] : on quitte le répertoire « /home/unknow/PongoOS/newlib/build/libc/ctype »
Making all in search
make[5] : on entre dans le répertoire « /home/unknow/PongoOS/newlib/build/libc/search »
make[5]: rien à faire pour « all ».
make[5] : on quitte le répertoire « /home/unknow/PongoOS/newlib/build/libc/search »
Making all in stdio
make[5] : on entre dans le répertoire « /home/unknow/PongoOS/newlib/build/libc/stdio »
make[5]: rien à faire pour « all ».
make[5] : on quitte le répertoire « /home/unknow/PongoOS/newlib/build/libc/stdio »
Making all in string
make[5] : on entre dans le répertoire « /home/unknow/PongoOS/newlib/build/libc/string »
make[5]: rien à faire pour « all ».
make[5] : on quitte le répertoire « /home/unknow/PongoOS/newlib/build/libc/string »
Making all in signal
make[5] : on entre dans le répertoire « /home/unknow/PongoOS/newlib/build/libc/signal »
make[5]: rien à faire pour « all ».
make[5] : on quitte le répertoire « /home/unknow/PongoOS/newlib/build/libc/signal »
Making all in time
make[5] : on entre dans le répertoire « /home/unknow/PongoOS/newlib/build/libc/time »
make[5]: rien à faire pour « all ».
make[5] : on quitte le répertoire « /home/unknow/PongoOS/newlib/build/libc/time »
Making all in locale
make[5] : on entre dans le répertoire « /home/unknow/PongoOS/newlib/build/libc/locale »
make[5]: rien à faire pour « all ».
make[5] : on quitte le répertoire « /home/unknow/PongoOS/newlib/build/libc/locale »
Making all in sys
make[5] : on entre dans le répertoire « /home/unknow/PongoOS/newlib/build/libc/sys »
Making all in .
make[6] : on entre dans le répertoire « /home/unknow/PongoOS/newlib/build/libc/sys »
[... cut ...]
      start$l0 in entry-54fb00.o
  "_enable_interrupts", referenced from:
      __task_switch_asserted in int-104268.o
      _task_load_asserted in int-104268.o
  "_sep_check_mailbox", referenced from:
      _sep_racer in sep_racer-72949f.o
  "_sep_fast_check_mailbox", referenced from:
      _sep_racer in sep_racer-72949f.o
ld: symbol(s) not found for architecture arm64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make: *** [Makefile:99 : build/Pongo] Erreur 1

slayy2357, 2024-07-11:
> Ok, try adding this before your `make` invocation:
> 
> ```
> EMBEDDED_LDFLAGS="-fuse-ld=$(which ld64) -Wl,-lto_library,$(llvm-config --bindir)/../lib/libLTO.so"
> ```

OK, so, on ubuntu it didn't work, same error. But on my main distro, kali, it work with your argument before make invocation. I had the same error on both distros so at least it works on Kali linux. Thanks

Siguza, 2024-07-11:
Hmm, that's odd. I've been meaning to look into LTO issues, but got stuck on trying to cross-compile a dependency for ld64... one of these days I'll hopefully have time to get back to that.

closed 2024-07-11 by slayy2357