checkra1n/PongoOS #187: [Question] Can pongoOS access to non-volatile memory (like flash) ?

slayy2357, 2024-07-25:
I was wondering if it's possible to directly access non-volatile memory (such as flash) on supported devices with pongoOS. Specifically, I'm curious to know if pongoOS allows operations like reading from or writing to non-volatile memory regions, and if so, what methods or tools are available to achieve this ?

Siguza, 2024-07-25:
This would require a NAND/NVMe driver, which we currently don't have.

slayy2357, 2024-07-25:
> This would require a NAND/NVMe driver, which we currently don't have.

I'm curious about how it is possible to obtain or develop a NAND/NVMe driver for pongoOS. Would it involve writing the driver from scratch, or are there existing sources or open-source projects that it is possible to adapt for this purpose ?

Siguza, 2024-07-25:
Asahi Linux has a driver, but I don't know how close the hardware of M1 and later is to A11 and older in that regard. I don't know how central of a role ANS2 plays, etc.

But if that fails you, Project Sandcastle must've implemented a driver as well, and that targets primarily A10, IIRC?

slayy2357, 2024-07-25:
> Asahi Linux has a driver, but I don't know how close the hardware of M1 and later is to A11 and older in that regard. I don't know how central of a role ANS2 plays, etc.
> 
> 
> 
> But if that fails you, Project Sandcastle must've implemented a driver as well, and that targets primarily A10, IIRC?

Ok, and given that Project Sandcastle is designed for A10 chips, is it theoretically possible to port the driver to an A8 device, for example?

Siguza, 2024-07-25:
Don't think so. A9 and newer use NVMe, A8(X) and older use ANC (Apple NAND controller).

slayy2357, 2024-07-25:
> Don't think so. A9 and newer use NVMe, A8(X) and older use ANC (Apple NAND controller).

Ok, thanks anyway

closed 2024-07-25 by slayy2357