checkra1n/PongoOS #189: Build fails on macOS

UnCor3, 2024-08-23:
MacOS Sonoma 14.0 i386 (non M) Xcode 15.4 | MacOSX14.5.sdk | iPhoneOS17.5.sdk

Build fails for some reason I've specified all versions above 

I have tried doing `make -C newlib distclean; make -C newlib all` but didn't help

Any help is greatly appreciated 

[image]

Siguza, 2024-08-23:
I think this means you're building the `master` branch... do you actually want that, or do you wanna build the `iOS15` branch? It shouldn't happen there.

Either way, `EMBEDDED_CFLAGS=-Wno-error` should work for the master branch.

UnCor3, 2024-08-23:
Never realized that I cloned the master branch. After cloning iOS15 branch and running make all worked fine thank you

closed 2024-08-23 by UnCor3