checkra1n/PongoOS #190: how to load Linux

UnCor3, 2024-08-23:
Hey I've been able to boot to pongoOS on my iPhone 6s. Using the makefile located in scripts I built pongoterm and it works fine I was able to communicate with the device. Then I wanted to load Linux , created a venv and installed pyusb then I got the stable Linux kernel from [here](https://kernel.org/) (lmk if I need a specific kernel) finally I did `python3 load_linux.py -k /path/to/extracted/kernel` but I got this error `no dbtpack specified` I do not have any knowledge about dbtpack can you help ?

[image]

UnCor3, 2024-08-23:
Update, tried [these](https://github.com/SoMainline/linux-apple-resources/blob/master/HOWTO.md) steps but didn't work

Siguza, 2024-08-28:
Linux support in upstream PongoOS hasn't been touched in quite a while. I know other people have experimented with forks, but I don't know what the last status of that is.

I know @konradybcio had a fork of PongoOS 2.5.x but porting that to 2.6.x ran into some memory corruption issues.

And I think @asdfugil has been working on getting m1n1 working?

asdfugil, 2024-08-28:
Load linux in checkra1n/PongoOS is really only for loading sandcastle, it never worked in pongo 2.6.

The part of the linux code in pongoOS that is actively being used is the lzma decompression code, for unrelated purposes in palera1n/PongoOS to allow pongoOS to receive LZMA compressed files (reduces jailbreak executable size).

Loading linux from m1n1 is being worked on and only works properly on A9-A10(X). A7-A8(X) has some data abort issues as well as `legacy console [tty0] enabled` (didn't actually enable) and A11 is very weird, it could have the same problem as A7-A8(X) sometimes, have problems with secondary CPUs instead, or works with no clear reason as to why. 

Loading m1n1 is enter at highest exception level, disable SRAM, and jump to 0x800 into the image, x0 = pointer to BootArgs struct.
https://github.com/asdfugil/pongoOS/tree/mini

It is unlikely that loading linux from PongoOS will be considered anymore, since it does not support handling secondary CPUs. Plus given that porting linux now is basically hacking on asahi linux to work here it is probably a wise idea to stay close to that bootchain.